travisgoodspeed/md380tools #628: MD-390 Extended Retry

k6ken, 2016-12-05:
The MD-390's I've just acquired do through endless retries when trying to get a talk permit from a DMR repeater. So, if I am out of range of a DMR repeater and I bump the PTT key, the radio tries endlessly to get the talk permit, and you can't change channels. I loaded the new experimental firmware but noticed there was no change.

  Is there a way in the firmware to ad an option to the tools menu that controls how many retries the radio will go through before it bonks/gives a talk deny tone and quits trying? This would be a cool added feature. 

https://groups.google.com/forum/#!forum/md380tools

travisgoodspeed, 2016-12-11:
Can anyone reporduce @k6ken's issue?

--Travis

travisgoodspeed, 2016-12-14:
Closing issue.  Re-open if we can confirm the problem and reproduce it.

closed 2016-12-14 by travisgoodspeed