travisgoodspeed/md380tools #634: User file display

PD1LOI, 2016-12-11:
If the user can not be seen, it is then possible to display the screen as original again?
[image]

PD1WP, 2016-12-11:
We fully agree with the desire of the PD1LOI to adjust the RX screen like the picture. But with retention of the entire database. The current RX screen is now half empty and this will not be turned back. This could be implemented as soon as possible? Thanks for the hard work . greatings PD1WP Wilfred

sijskes, 2016-12-11:
this is not a forum. go to the mailinglist.

travisgoodspeed, 2016-12-11:
Hey @PD1LOI and @PD1WP,

It's really not cool for you to keep starting feature requests and conversations as Github issues.  Issues are *only* for features and bugs which are under active development.  They are not for ideas, complaints, or advice.  They are not a forum.

Each one of these issues gets added as a TODO list for the project developers, and it sucks the fun out of our project for that list--and our inboxes!--to keep filling up.  Please, *please* take stuff like this to the mailing list and stop creating Github issues.

https://groups.google.com/forum/#!forum/md380tools

73 from Philadelphia,
--Travis KK4VCZ

closed 2016-12-11 by travisgoodspeed