travisgoodspeed/md380tools #636: please rebase pull for direct push

sijskes, 2016-12-13:
We have removed a file from the repo, and we don't want it re-commited.

So we ask of all collaborators to do a 'git pull --rebase' in order to clean their local repo as well.

sijskes, 2016-12-13:
And thanks to Jamie from github support!

travisgoodspeed, 2016-12-14:
One last notice: Please run `git pull --rebase` to clean up local repositories after some recent repo surgery!

closed 2016-12-14 by travisgoodspeed

ispkorte, 2016-12-15:
how can we do it ?
`git pull --rebase` make an error 

need before
```
  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"
```
and then ?

sijskes, 2016-12-15:
please be more specific about the error.

ispkorte, 2016-12-16:
You are not currently on a branch. Please specify which
branch you want to merge with. See git-pull(1) for details.

    git pull <remote> <branch>