travisgoodspeed/md380tools #637: Ubuntu 16.10 "Device not found"

ryanturner, 2016-12-14:
Hi,

I've successfully built the md380tools on ubuntu 16.10, and I copied the udev rule, but I can't seem to get the device found. I do have it in DFU mode, alternative red and green LED colors (MD380). 

`/md380-dfu time
Device not found
`

lsusb doesnt show the device and udevadm monitor shows no activity. Should i investigate the programming cable and device, or is there another troubleshooting step I should take?

travisgoodspeed, 2016-12-14:
Almost certainly the cable or the radio hardware is to blame; this isn't a software issue.  I feel your pain, as my Hytera radio and cable suffer the same problem.

Closing issue as out of scope.

closed 2016-12-14 by travisgoodspeed

ryanturner, 2016-12-14:
@travisgoodspeed OK -- should I expect to see an entry in lsusb?

travisgoodspeed, 2016-12-14:
Yup, it should always show up there.

ryanturner, 2016-12-14:
Cool. I updated the radio's firmware in the past, but with a home made cable, and since I've disassembled that. Bought a TYT programming cable, sounds like it's a dud. Thanks for your help @travisgoodspeed