travisgoodspeed/md380tools #642: Boot Chime + Other Sounds

pw721, 2016-12-23:
The new MD-398+MD-680 seem to utilize a new boot chime that is the same as the motorola sound (linked below). I find the current MD-380 sounds to be quite bad. Wondering if the MD-380,390 could be upgraded with new sounds. 

Sounds:
Boot Chime - https://www.freesound.org/people/wiredpig/sounds/94386/
Keypad Click - https://www.freesound.org/people/wiredpig/sounds/94385/
PTT - https://www.freesound.org/people/wiredpig/sounds/94388/

sijskes, 2016-12-23:
@pw721 are you going to implement it? have you read the prerequisites for filing an issue? closing. reopen if you yourself are going to implement it.

closed 2016-12-23 by sijskes

pw721, 2016-12-23:
The problem is I do not know how to implement it,.

travisgoodspeed, 2016-12-23:
We're sympathetic, but we haven't got time to implement every feature that folks would like.  There's a whole bundle at #370 that we'll never have time for, and we can't fill the issue tracker with feature requests.

You might try asking on the forum or in the IRC channel.

73 from New York,
--Travis KK4VCZ

sijskes, 2016-12-23:
@travisgoodspeed @pw721 and the sound system is barely known.