travisgoodspeed/md380tools #643: Time sync.

geiraage, 2016-12-26:
## Want to have issue!

Is it possible to have a "make" command that sync time from NTP or computer using local timezone?

Now you need to do it manually, or from the TYT program.

Merry christmas.

claywalker0929, 2016-12-26:
I am willing to work on this.

travisgoodspeed, 2016-12-26:
You neighbors are in luck, as @ea already implemented this feature in #21.  This is done through the md380-dfu tool, with usage below.

```
dell13% ./md380-dfu 

Usage: md380-dfu <command> <arguments>

Print the time from the MD380.
    md380-dfu time

Set time and date on MD380 to system time.
    md380-dfu settime
```

@claywalker0929 If you'd like add support for specifying UTC or the local timezone, we'd happily accept such a pull request.

73 from Reno, Nevada,
--Travis KK4VCZ

closed 2016-12-26 by travisgoodspeed

geiraage, 2016-12-26:
Tx :-)