travisgoodspeed/md380tools #644: Decoding analog data transmissions

geiraage, 2016-12-27:
I have been experimenting with the md380-tool dmesgtail command. And from digital channel, you can see most of the information of the call. 

This brings me to this. 

Is there any possible decode well know analog data transmissions?

I would be very cool, if possible to be able to monitor and decode content for example.

**Maritim Digital selective calling** or **DSC** that sends always on VHF 156,525 (Maritim Ch 70)

**Marine Automatic Identification System (AIS)** thats using VHF Channel A 161.975Mhz (87B) 
Channel B 162.025Mhz (88B)

**Automatic Packet Reporting System (APRS)** 

Sample APRS VHF frequencies
* 144.390 MHz – Colombia, Chile, Indonesia, Malaysia, North America, Thailand
* 144.575 MHz – New Zealand [5][6]
* 144.660 MHz – Japan
* 144.800 MHz – South Africa, Europe,[Russia
* 144.930 MHz – Argentina, Uruguay
* 145.175 MHz – Australia
* 145.570 MHz – Brazil
* 430.5125 MHz – Netherlands (UHF)
* 433.800 MHz Primary - Europe (UHF)
* 432.500 MHz Secondary - Europa (UHF)

travisgoodspeed, 2016-12-27:
Any volunteers to figure out raw audio sampling in analog mode?  If not, I'll move this to #370.

The issue tracker is not a place for feature requests.  Please send those to the mailing list or the IRC channel.

travisgoodspeed, 2017-01-07:
Moved feature request to #370.

closed 2017-01-07 by travisgoodspeed