travisgoodspeed/md380tools #645: Make Error In Cygwin

gomker, 2016-12-27:
This issue looks to be the same as Issue #63 

	"make" -f Makefile_orig unwrapped/D002.032.img
	make[5]: Entering directory '/home/dlane/repo/md380tools/firmware'
	../md380-fw --unwrap bin/D002.032.bin unwrapped/D002.032.img
	DEBUG: reading "bin/D002.032.bin"
	Traceback (most recent call last):
	  File "../md380-fw", line 7, in <module>
		main()
	  File "/home/dlane/repo/md380tools/md380_fw.py", line 185, in main
		md.unwrap(input)
	  File "/home/dlane/repo/md380tools/md380_fw.py", line 114, in unwrap
		header = header.unpack(img[:256])
	struct.error: unpack requires a string argument of length 256

Looking at my `/firmware/bin` I see that `D002.032.bin` is 0 bytes

My environment is setup using this [article](https://learn.adafruit.com/tytera-md-380-dmr/updating-md-380-firmware) as a guide

travisgoodspeed, 2016-12-27:
Please post a log of `make distclean dist`, built while your machine is online.

gomker, 2016-12-27:
No errors on that output - I did not realize `distclean` was a build option, I was running `make clean dist` 
Should have read the Make file a bit more.

Thanks

travisgoodspeed, 2016-12-27:
@gomker No worries, this is a real bug that we ought to fix.  An explanation follows.

## Explanation

What likely happened was a network error as @gomker was initially fetching the firmware image, giving him a zero-byte file.  Users had reported this in the past, and for a while, 'make clean' would erase the temporary files.  This was great for users who were online, but meant that 'make clean all' would fail when away from the Internet.

We can fix this by adding a 'make verify' stage to the `firmware/` directory, or by committing the firmware directly to the repo.

## Resolution

Close this issue when there is no longer danger of a corrupted `firmware/` directory, but also that the Internet is only required for at most the initial build.

KG5IEN, 2017-05-23:
This can be closed now, right?  Or does more work remain?

closed 2017-05-27 by travisgoodspeed