travisgoodspeed/md380tools #648: promisc failing or needs documentation

ZeroChaos-, 2016-12-31:
md390, no gps

I have configured my radio for the local repeater, and when I turn on promisc (with or without users.csv on) it doesn't seem to be giving me the display window with color code anymore.  In reality, it seems that the feature makes no difference at all in my display.  Does the promisc window simply not show up if the radio is already properly configured for what is being received?  If that is the case, I can work on some light documentation for the menu features to help prevent bugs like this from being filed for intended behavior.

travisgoodspeed, 2017-01-07:
@ZeroChaos- I'm trying to reproduce your issue, but it seems to work as expected in firmware from November 29th and today, 7 January patched against D13.020.

Which firmware version are you running?  Do you hear the audio?

ZeroChaos-, 2017-01-08:
I just did "make flash".  Radio info just shows md380tools ver 2016-12-31 but no other numbers or anything.  I do seem to hear random talk groups, but when I tested a build in September (also built by me in the same way) it seemed to show color code and now it is not.  Apologies for the sporadic testing.  Until next week I only have one radio and I am out of range of anything to monitor while at home :-)  After shmoocon I'll have two and be a bit more useful.

travisgoodspeed, 2017-01-08:
The color code isn't supposed to be shown on the incoming call screen.  This is what it should look like, showing the call information from the local database.

[image]

If you need to see the color code, you can enable the Developer Mode through the MD380Tools menu and then hit the 9 button to see it.  (Press 7 to return to the normal screen or 8 to see other information.)

See you at Shmoo,
--Travis

ZeroChaos-, 2017-01-08:
A few things.

First, oh man that is AWESOME.  I had no clue how to use dev mode and that's incredibly useful looking.  Thanks.

Second, I swear that color code used to be shown in the promisc screen (or I hallucinated it in a dream) and it looks like there is enough room.  Can we please show that in promisc?

travisgoodspeed, 2017-01-08:
It might have been shown in the early days of the screen, but it's not particularly useful as we don't have promiscuity relative to CC or TS, but only for the destination LLID (Talkgroup or User.)  I don't think we'll be bringing it back to that screen.

If we manage to figure out timeslot or color code promiscuity, then we'll add it to the bottom of that display. 

Closing issue.

closed 2017-01-08 by travisgoodspeed