travisgoodspeed/md380tools #649: AppVeyor is Broken Again

travisgoodspeed, 2017-01-07:
AppVeyor is failing again, this time because of perl issues.  (Do we really depend upon Perl?)

```
StrawberryPerl v5.20.1.1 [Approved] - Possibly broken
strawberryperl package files install completed. Performing other installation steps.
Downloading StrawberryPerl 64 bit
  from 'http://strawberryperl.com/download/5.20.1.1/strawberry-perl-5.20.1.1-64bit.msi'
Progress: 100% - Completed download of C:\Users\appveyor\AppData\Local\Temp\1\chocolatey\StrawberryPerl\5.20.1.1\strawberry-perl-5.20.1.1-64bit.msi (72.78 MB).
Download of strawberry-perl-5.20.1.1-64bit.msi (72.78 MB) completed.
WARNING: Missing package checksums are not allowed (by default for HTTP/FTP, 
 HTTPS when feature 'allowEmptyChecksumsSecure' is disabled) for 
 safety and security reasons. Although we strongly advise against it, 
 if you need this functionality, please set the feature 
 'allowEmptyChecksums' ('choco feature enable -n 
 allowEmptyChecksums') 
 or pass in the option '--allow-empty-checksums'. You can also pass 
 checksums at runtime (recommended). See choco install -? for details.
The integrity of the file 'strawberry-perl-5.20.1.1-64bit.msi' from 'http://strawberryperl.com/download/5.20.1.1/strawberry-perl-5.20.1.1-64bit.msi' has not been verified by a checksum in the package scripts.
Do you wish to allow the install to continue (not recommended)?
[Y] Yes [N] No (default is "N")
  Confirmation (`-y`) is set.
  Respond within 30 seconds or the default selection will be chosen.
ERROR: Empty checksums are no longer allowed by default for non-secure sources. Please ask the maintainer to add checksums to this package. In the meantime if you need this package to work correctly, please enable the feature allowEmptyChecksums, provide the runtime switch '--allow-empty-checksums', or pass in checksums at runtime (recommended - see 'choco install -?' / 'choco upgrade -?' for details). It is strongly advised against allowing empty checksums for non-internal HTTP/FTP sources.
The install of strawberryperl was NOT successful.
Error while running 'C:\ProgramData\chocolatey\lib\StrawberryPerl\tools\ChocolateyInstall.ps1'.
 See log for details.
Chocolatey installed 0/1 packages. 1 packages failed.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
Failures
 - strawberryperl (exited -1) - Error while running 'C:\ProgramData\chocolatey\lib\StrawberryPerl\tools\ChocolateyInstall.ps1'.
 See log for details.
Command exited with code -1
```

Close this issue when AppVeyor is either fixed or removed.  (Travis-CI seems to be far more stable.)

m0ivq, 2017-01-08:
The Appveyor build image was recently updated. Have edited the config file to revert to the previous image until I can fix the issue properly.

As AppVeyor CI is only used for Windows, do we continue using AppVeyor and wait for the next time it breaks or do we fix Windows (GnuWin32) build issues if/when they are reported and rely solely on Travis-CI?

travisgoodspeed, 2017-01-08:
I'd like to either get proper Windows support with good documentation or scrap Windows entirely and ask those users to grab binaries rather than building from source.

Toward proper Windows support, we might build in MSYS2, the modern replacement for Cygwin.  It has a proper package manager and more recent packages, but it is currently broken on AppVeyor.
https://github.com/appveyor/ci/issues/733

Toward scrapping it, we would need to make binary releases.  Toward that end, we have a server sitting idle at md380.org which could host regular releases.

Any thoughts, @m0ivq?

m0ivq, 2017-01-08:
@travisgoodspeed I haven't used MSYS2 myself (though have used Cygwin) but I think supporting a single Windows solution is a good idea. Most Windows only users probably use pre-built binaries but there are still many who would still like the option of building themselves. One of the main features used is the user database which currently wouldn't be available if just using pre-built binaries.

travisgoodspeed, 2017-01-08:
Okay, we'll keep Windows by try to narrow it to a single style.  I'll try to get it going in MSYS2 this afternoon and keep notes.

In the IRC channel today, issue #186 cropped up again.  Maybe flashing that database from Windows is as important as compiling the code?

--Travis

travisgoodspeed, 2017-01-17:
Broken again.

m0ivq, 2017-01-17:
@travisgoodpeed unless someone else can look into it fairly soon you may wish to disable Appveyor for the time being. I'm going to be away from a computer for a while due to other commitments. I've just checked the Appveyor updates page and unless they haven't updates the page yet, the build worker image hasn't changed.

travisgoodspeed, 2017-01-21:
[image]

Okay, I'll cut Appveyor this afternoon.

travisgoodspeed, 2017-01-21:
I've disabled the webhook.  Closing issue.

closed 2017-01-21 by travisgoodspeed

m0ivq, 2017-01-21:
I'll look into working out how to do automated builds on commit/pull request on my own server in MSYS2. Once I have a good enough understanding I'll open a separate issue for tracking.

m0ivq, 2017-01-30:
#675