travisgoodspeed/md380tools #655: TYT MD 390 VHF - flashdb can't program spy flash wrong flash type

NQ4Y, 2017-01-14:
Using TYT MD 390 and KD4Z VM. 
The flash command worked flawlessly to update firmware.
MD380Tools Ver 2017-01-12
CP Ver. : V01.36

The flashdb command returns this message:
Record Count 51629 /home/tyt/user.bin
SPI Flash ID: ef 40 14
W25Q80BL 1MByte
can't programm spi flash wrong flash type

thoughts?

thank you,

Terry NQ4Y

KD4Z, 2017-01-14:
Terry,

It appears that your radio only has 1 Meg of flash space, and can't take
the UsersCSV data.  I've see this on the VHF MD-380 radios.
Typical radios have 16 Meg flash space.

You can enjoy the other features of the firmware, but you will need to
keep adding users in your Codeplug contacts area.

Warren, KD4Z

On 1/13/2017 8:24 PM, NQ4Y wrote:
> Using TYT MD 390 and KD4Z VM.
> The flash command worked flawlessly to update firmware.
> MD380Tools Ver 2017-01-12
> CP Ver. : V01.36
>
> The flashdb command returns this message:
> Record Count 51629 [...]

NQ4Y, 2017-01-14:
Hi Warren,
  Thank you so much for the info!

-Terry NQ4Y

closed 2017-01-14 by NQ4Y