travisgoodspeed/md380tools #656: Menu Translation

amvidalrc, 2017-01-14:
Hello. Im trying to find the string of the original firmware to translate to Spanish. I have found the md380tool menu, but I want to know if I can modify the original strings of TYT fw. ¿Where can I find them? I want to translate to Spanish and German. 

BR

geiraage, 2017-01-14:
I can do Norwegian,  Swedish and Danish.

14. jan. 2017 1.49 p.m.

travisgoodspeed, 2017-01-14:
`rabin2 -zzq foo.img` identifies all strings in the image, including the UTF16 that the firmware likes so much. Add 0x0800c000 to get the address in device memory.

amvidalrc, 2017-01-15:
What about the possibility of adding a language options on MD380Tools menu with instant change of strings? With rabin2 only get unreadable str's.

Wolf-DL4YHF, 2017-01-15:
Keep in mind that we only have a limited amount of memory in the precious *internal* Flash (not the SPI flash)... having multiple languages in there may be wasteful, when only one of them is only ever used.
Maybe better to have a string table in the external flash (SPI), if that's not too much effort.

travisgoodspeed, 2017-01-15:
Self-assigning.

I've begun code for static localization, which will produce firmware images for western languages.  I'll spin up a separate issue for each language translation after the localization code is stabilized.

paulomartinspt, 2017-01-16:
i can translate to Portuguese (PT-PT)

IK8JHL, 2017-01-17:
I' m ready for italian

travisgoodspeed, 2017-01-21:
Howdy y'all,

We are now ready for pull requests to add language support.  Please send pull requests against /locales/strings.txt with the English line first, followed by your localized language.  An example of the "Unprogrammed" error is provided in my yankee attempts at Hungarian, French, and German.

The localizing tool works, but it's very ugly C and ought to be rewritten before we distribute localized firmware.  It does not provide safety checks that the new string is as short as the old.

Do not use accent marks, as the font has no support for them.  If you must, include the accent-marked string as a comment above the accent-less string.

73 from New York,
--Travis

IK8JHL, 2017-01-21:
Sorry Travis im not sure to understood, must to write in strings.txt somthing like this ? : 
# 0xc53c0 28 13 Delete Entry?
it Cancellare?
# 0xc53dc 26 12 Radio Enable
it Abilita radio
# 0xc53f8 28 13 Radio Disable
it Disabilita Radio
# 0xc5414 26 12 Edit Number:
it Edita Numero
# 0xc5430 26 12 Text Message
it Messaggio di Testo

and after send to you?

travisgoodspeed, 2017-01-21:
Yup, that'll do just fine.

travisgoodspeed, 2017-01-21:
Some German strings have been added with #661.  See also 0044581, which adds matching English strings so that it will build.

IK8JHL, 2017-01-21:
Sorry again for my inexperience , I have add parts of italian language  in my file strings.txt but where i can send?

travisgoodspeed, 2017-01-21:
No worries, neighbor.  If you email it to me, Travis at Radiant Machines dot com, I'll take care of merging it for you.

SlashBuster, 2017-01-23:
Do I have to run make dist in the locales directory?

Thank you in advance
DL3KB

travisgoodspeed, 2017-01-23:
Localization is not yet handled by the Makefiles. That might be added this weekend.

SlashBuster, 2017-01-23:
So how can I compile it?

travisgoodspeed, 2017-01-23:
I pushed a proper makefile for localizing D013.020, which is the non-GPS firmware, to DE, HU, IT, and ES.  Run 'make clean flash-de' to flash the German firmware to your radio.

Please note that this is very experimental, and almost certainly some of the strings are illegally long, overwriting adjacent strings in memory.  Run at your own risk.

SlashBuster, 2017-01-23:
Thank you very much Travis. Just to make sure. 2 byte are in this case 8 caracters?

DL3KB

IK8JHL, 2017-01-25:
Something i can not clear for me , I enter directory locales , I run make clean flash-it , i write the radio .When the radio go on have write  the Patched firmware not original with new language . I have tryed with flash-de and also flash-es  Wath i do wrong ?

travisgoodspeed, 2017-01-25:
The current Makefile only produces patched firmware.  You can edit the Makefile to make unpatched firmware, but it's not the goal of this project to support a la carte patching.

IK8JHL, 2017-01-25:
Ok Travis , i understand , but the language inside dont change its only English , if write for example flash-es i mast have Spanish language inside or not ?

travisgoodspeed, 2017-01-25:
Apologies, I was only testing Hungarian, and then only one word.  'make clean flash-es' ought to work now, though perhaps with bugs where strings are too long.

IK8JHL, 2017-01-25:
The only language that i dont have tested ;-) Hi Hi .Now its all clear . I testing with Hungarian for now , i want test the max number of characters . .......i continue the work

IK8JHL, 2017-01-25:
But reading in string.txt  i view many translete words exceding lenght of words, example , maybe its not good 
# 0xc544c 26 12 Message Send
en Message Send    (12 OK)
de SMS gesendet    (12OK)
es Mensaje enviado (15 no)
it Mess.Inviato   (12OK)

travisgoodspeed, 2017-01-25:
Yes, these long strings will cause some trouble.  We will need to abbreviate them.

IK8JHL, 2017-01-25:
Ok when you are ready i can test italian , for correct lenght of words

dragstor, 2017-03-07:
Could it be possible that those long strings cause merging of specific words in menu items?

To be precise, "Mensaje enviado" should be 12 chars long, whereas it's 15. So, could this string cause issues with, let's say, Serbian translation?

I'll attach screenshots when I get home from work :) 

Thanks!
73

IZ4FXC, 2017-03-14:
Hi everybody!
Would someone please tell me where am I wrong?
In attachment the pic with my locales folder and the error with "make clean flash-it" command.
Thanks in advance, Greg.
[image]

travisgoodspeed, 2017-03-14:
@IZ4FXC It looks like you don't have a C compiler installed in MinGW.

IZ4FXC, 2017-03-14:
Hi Travis,
      First of all thank you for your attention.
      This is my MinGW installation, is there anything missing?
      Greg.
    
    Il 14/03/2017 15:30, Travis Goodspeed
      ha scritto:
    
    
      @IZ4FXC It looks like you don't have
        a C compiler installed in MinGW.
      —
          it on GitHub, or mute
          the thread.

travisgoodspeed, 2017-03-14:
This is not a support forum. Please ask about the proper MinGW environment on the forum or the mailing list.

IZ4FXC, 2017-03-14:
I apologize, many thanks. :)
    Gregorio De Michele
Assistenza informatica AUSL Modena
V.le Martiri 56
41026 Pavullo N.F. (MO)
tel. 0536 29247
cell. 320 2260953
Orari ufficio.

    Il 14/03/2017 16:20, Travis Goodspeed
      ha scritto:
    
    
      This is not a support forum. Please ask about the proper MinGW
        environment on the forum or the mailing list.
      —
          it on GitHub, or mute
          the thread.