travisgoodspeed/md380tools #670: [FEATURE] Vocalization

paulomartinspt, 2017-01-24:
Throw at Ye Grande Ol' Wish List for Christmas

A friend asked me to post this here.
It would be great for nonvisual to have vocalization of the menu options, like it happens with some radios options/channels. I know we have limited memory and functionalities but we never thought the MD380/390 would do all the things you guys make them do nowadays... 👍 
Something for you guys to think about.
Thanks for the attention.

travisgoodspeed, 2017-01-24:
Closing as a duplicate of #580.

We'd love to help visually impaired hams, but for now, it's probably best that they use a radio which already has audio vocalization, such as the Tytera MD280.

73 from Pizza Rat City,
--Travis KK4VCZ

closed 2017-01-24 by travisgoodspeed

paulomartinspt, 2017-01-26:
Sorry for the duplicate Travis...
Thanks for the reply