travisgoodspeed/md380tools #671: /usr/bin/python2 not createable in rootless mode

deBaer, 2017-01-26:
In the rootless mode of macOS Sierra, it is not possible to create the link from /usr/bin/python2 as described in the README. Deactivating rootless mode is strongly discouraged. You can however create /usr/local/bin/python2, but the Makefile needs /usr/bin/python2 to exist.

gianfrancop, 2017-02-06:
I did this:

1. disable rootless
2. create the symlink
3. go back to rootless
 
(I also use no-rootless to delete unused languages and other non-conventional stuff btw)

svaio, 2017-03-06:
Or ... 

####Fix the environment statements for  OS X > 10.10:####

If you are using OS X > 10.10, the /usr/bin filesystem is looked by the "rootless" feature. Therefore you cannot create a lnk to python2 in this directory.

What helps and also solves an error with the pyusb Library is to run brew and build your won python2.7 package:

    brew install python
    brew install libusb
    pip install pyusb

After this, `make flash` as expected.

PR #899 (2019-03-23): Update Mac OS X instructions
Fixes #671, #898.
commit 6a5a92e9 (2019-03-23): Update Mac OS X instructions

Fixes #671, #898.
closed 2019-03-23 by mach327