travisgoodspeed/md380tools #681: Talker Alias

w7xm, 2017-02-12:
#163 was closed and would like to see this reopened and actively worked on since Brandmeister is now supporting the Talker Alias along with openSPOT and MMDVM. Hytera is also supporting this with their latest firmware updates. I have a request into Motorola to see when they might support this in their future firmware releases and will report back when they do. But this should not inhibit from having the MD380 now using it since Brandmeister is using it.

travisgoodspeed, 2017-02-14:
We're aware that folks would really like this feature, but for now, no one on the dev team has volunteered to write it.  Pull requests are always welcome, of course.

Closing as a duplicate of issues #163 and #370.

zarya, 2017-02-17:
First working version of Talker alias in netmon
[image]

travisgoodspeed, 2017-02-17:
Reopening issue now that progress is being made.  Thanks, @zarya!

lb5je, 2017-02-20:
Is it only Hytera radios that generate this info. Seems that not all users "send" TA.

73 de LB5JE

w7xm, 2017-02-20:
This is true. Hytera is the only one presently Sending the TA info.
Otherwise it is generated by Brandmeister if no ta is sent.

jruys, 2017-02-27:
I notice the MD380 displays TA from Brandmeister OK, but does not show TA when receiving directly from a Hytera MD/PD 785. I'm using compressed IP and tried both 7, 8 and 16 bit encoding. 

Is this expected behavior? Hytera seems to send the TA later than Brandmeister does.

DL2MF, 2017-03-17:
Via our local repeater TA from Hytera PD785 is received during first TX ans already shown at the end of the first RX_screen change in my fork. Did you already try this?

shawnchain, 2019-12-26:
Hi,
I would like to know if MD380 could decode the TA info from Brandmeister or not ?

I did have "Talker Alias/TA & UserDB" options in the "Show Calls" menu in MD380 but the screen always displays "No entry in your user.csv" error message instead of the desired TA string.

But another Hytera PD980 works fine for the same RF signal.

Am I missed something ? 

Thanks,

Shawn

PD0DIB, 2019-12-26:
For support:
See / search for theTYT MD380 tools on Facebook...
73 de PD0DIB | Rob

DL2MF, 2020-01-06:
Why should someone interested in developer informations visit an F***book group (people aware of data protection might not use it), when developers can be reached and reply here directly?

TA and UserDB are two complete different features:
- talker alias is digital information (name/callsigm/user info) set and transmitted by the user during his DMR transmission
- UserDB is an index file of DMR id's where a lookup is being mad with the DMR id submitted from the repeater and displaying the textinformation from the DB if the ID was found.

rogerclarkmelbourne, 2020-01-07:
@DL2MF  +1

F**book is another form of dark web, with very limited visibility to search engines etc.

travisgoodspeed, 2020-01-07:
I hate Facebook as much as the rest of you, but I really can't handle the noise of feature requests going directly into the Github issue tracker.

Closing as merged in #698.

closed 2020-01-07 by travisgoodspeed

shawnchain, 2020-01-10:
I'm not requesting the feature here, actually I have developed my own reflector server like BM and supports customized TA info. Yes it eats me lots  of time to read the 3 volumes of ETSI spec...
 
But unfortunately I do not have too much time dig into the MD380 code for the decoding issue currently. But I believe there is a bug there.
Anyway, thanks for the feedback.

Cheers,

BG5HHP

rogerclarkmelbourne, 2020-01-10:
@shawnchain 

I presume you've considered using the OpenGD77 for your tests. TA Rx and Tx are now supported