travisgoodspeed/md380tools #682: Annoying hum of modulation at latest firmware with back light dimmer

PD1WP, 2017-02-12:
Hello
I have just the latest firmware on my TYT MD-390 put but have a problem.
Since the firmware update with the dimmer is an annoying hum in the modulation.
The hum came after flashing the last firmware and never had them this in earlier versions.
If I go back a version is the hum away.
So he is really into the latest version.
Would you want to look for a solution please?
73ers Wilfred PD1WP

# PLEASE READ ME! #

When you click "Submit new issue" below, you are sending an email to
every developer of this project.  Our phones buzz, our computers beep,
and we are distracted from whatever we were doing.

Github Issues are not a discussion forum, and they are not a place to
share nifty ideas for new features or to ask for help.  Please discuss
those things on the mailing list, which is a forum that welcomes ideas
and discussion.

https://groups.google.com/forum/#!forum/md380tools

If you are certain that your bug is serious or a project developer has
agreed to implement a feature, then erase this text and replace it
with a clear explanation of exactly (1) what the problem is, (2) how
to reproduce the problem, and (3) what the expected behavior would be.
Please include as much detail as possible, such as whether the problem
is unique to our patches or also exists in the official firmware.
Make sure that your title accurately describes the problem.

PD1LOI, 2017-02-12:
Here exactly the same, hum audible analogue stations.

Wolf-DL4YHF, 2017-02-12:
If the hum is on transmit only, please try with "Backlt Hi" set to 
maximum. It's in the MD380Tools menu. The effect may be caused by the 
pulse-width modulation.
On both of my RT3's, I didn't experience such a problem.
We could try a higher PWM frequency, or use a default of 100 % duty 
cycle, i.e. full intensity.

Cheers,
   Wolf DL4YHF .

PD1LOI, 2017-02-12:
There are certainly MD 380-390 where it is not present, but my MD 380 without GPS New Vocode is evident.
I'm Testing.

PD1LOI, 2017-02-12:
As I Backlight High 9 (bright)  and backlight low to 0 (Off) still is the hum in analog signal present.

Wolf-DL4YHF, 2017-02-12:
Ok thanks - can you confirm that the dimmed backlight only affects your 
own transmission, but no hum on receive ?
I am looking for solution that switches to 'DC' (no PWM) for the 
backlight during transmit.
There must be a flag / variable somewhere that tells the current state 
of the PTT... any idea anyone ?
If such a flag exists, the solution would be trivial. Guess the problem 
depends on the internal resistance of the accumulator or power supply / 
regulator... when drawing 'constant current' during TX, it should be gone.

73, Wolf  DL4YHF .

PD1LOI, 2017-02-12:
When I have out dim function, you still hear buzzing with RX, TX out.

PD1WP, 2017-02-12:
I hear it in my RX. I do not know if my TX hum broadcast. If there are ongoing QSOs I hear contstant this low-frequency hum.
I can’t record this. I went back a firmware version and the hum is gone.
Wilfred PD1WP

Wolf-DL4YHF, 2017-02-12:
Ok Wilfred, thanks for the clear description. It's not necessary to 
record it.
When I press my RT3 to the ear, with the speaker volume cranked up, I 
can hear a tiny bit of hum myself but it's much weaker than the white 
noise (hiss) from the RX audio.
I have pulled a request to the md380tools main branch with a possible 
fix. With "Backlt Low" set to 0 (off), and "Backlt High" set to 9 
(brightest), there is no pulse-width modulation at all, as in the 
original firmware. Hopefully this fixes it, but of course radios 
affected by the hum will have no dimmed backlight during standby. And 
switching to max brightness just because the squelch opens (in FM) is 
not an option, at least not for me. The display is much too bright when 
used outdoors or in the car.

Cheers,
   Wolf .

p.s. we should move further descussions of this matter to the google 
group, to keep the github issue list noise-free. My apologies to the others.

travisgoodspeed, 2017-02-12:
Has this issue been fixed with the recent pull request merge?  If so, let's close the issue.

PD1WP, 2017-02-12:
No..  The hum is still there.   Sorry

Verzonden vanaf mijn Samsung Galaxy-smartphone.

PD1WP, 2017-02-12:
No the hum is still there.  I  makes a small recording

Verzonden vanaf mijn Samsung Galaxy-smartphone.

Wolf-DL4YHF, 2017-02-12:
Can others confirm the issue is still there, with the *latest* firmware 
from github ?
Again, and this is important:
"Backlt low" : set to ZERO in the md380tools menu.
"Backlt high" : set to NINE in the same menu.

In effect, this disables dimming, and I really cannot imagine what 
causes that hum if the backlight isn't "modulated" at all.
If the hum really originates from the backlight, it will have an audio 
frequency of exactly 104.65 Hz. Maybe you can measure it, and confirm 
that the frequency is not AC mains hum (50 or 60 Hz or harmonics).

Am 12.02.2017 um 17:06 schrieb PD1WP:
> No.. The hum is still there. Sorry

PD1WP, 2017-02-13:
Good morning  Wolf DL4YHF

De setting "Backlt = 0"
                   "Backlt = 9"
With these settings is the hum completely disappeared.
The dim function causes the hum.
If got a TYT mb-390 without GPS and the new vocoder
I hope that i give you enough information to your question.
Greetings and 73ers from Apeldoorn, the Netherlands
PD1WP
pd1wp.apeldoorn@gmail.com
My mobile phone number is +31614202011

Wilfred Pontfoort

Verzonden vanaf Samsung-tablet.

PD1WP, 2017-02-13:
I've tried various combinations of high and low. Once the dim function is switched on, the hum comes back.

Wolf-DL4YHF, 2017-02-13:
Ok, so I think we can close this issue. It's not a nice solution but 
it's viable.

We have to accept the fact that *some* radios cannot be dimmed due to 
unknown hardware reasons (poor voltage regulator ?), and we have a 
solution for those (by using the defaults "completely dark" or "maximum 
brightness").

Anything else, like dimming the backlight only if the FM-squelch is 
*not* open, would be much effort (and in most cases not necessary at all).

Cheers,
  Wolf .

travisgoodspeed, 2017-02-13:
Okay, closing the issue.

closed 2017-02-13 by travisgoodspeed

PD1LOI, 2017-02-14:
firmware-20170214-090001-4716c3f4.bin 
<https://pd0zry.nl/md380-fw/firmware-20170214-090001-4716c3f4.bin> This 
firmware does not hum anymore. Nor functe the maximum dim. For me the 
issue resolved

Thanks 73  PD1LOI