travisgoodspeed/md380tools #684: Backlight Timer issue in Unchanged

IK8JHL, 2017-02-13:
# PLEASE READ ME! #

When you click "Submit new issue" below, you are sending an email to
every developer of this project.  Our phones buzz, our computers beep,
and we are distracted from whatever we were doing.

Github Issues are not a discussion forum, and they are not a place to
share nifty ideas for new features or to ask for help.  Please discuss
those things on the mailing list, which is a forum that welcomes ideas
and discussion.

https://groups.google.com/forum/#!forum/md380tools

If you are certain that your bug is serious or a project developer has
agreed to implement a feature, then erase this text and replace it
with a clear explanation of exactly (1) what the problem is, (2) how
to reproduce the problem, and (3) what the expected behavior would be.
Please include as much detail as possible, such as whether the problem
is unique to our patches or also exists in the official firmware.
Make sure that your title accurately describes the problem.

travisgoodspeed, 2017-02-13:
Closing as an empty issue.

closed 2017-02-13 by travisgoodspeed

IK8JHL, 2017-02-13:
In new menu " Backlight tmr " i cannot select Unchanged 
if i select Unchanged , return to last setting ( 5 sec, 30 sec, or 60 sec