travisgoodspeed/md380tools #685: Backlight timer cant selected

IK8JHL, 2017-02-13:
In new menu " Backlight tmr " i cannot select Unchanged
if i select Unchanged , return to last setting ( 5 sec, 30 sec, or 60 sec )
# PLEASE READ ME! #

When you click "Submit new issue" below, you are sending an email to
every developer of this project.  Our phones buzz, our computers beep,
and we are distracted from whatever we were doing.

Github Issues are not a discussion forum, and they are not a place to
share nifty ideas for new features or to ask for help.  Please discuss
those things on the mailing list, which is a forum that welcomes ideas
and discussion.

https://groups.google.com/forum/#!forum/md380tools

If you are certain that your bug is serious or a project developer has
agreed to implement a feature, then erase this text and replace it
with a clear explanation of exactly (1) what the problem is, (2) how
to reproduce the problem, and (3) what the expected behavior would be.
Please include as much detail as possible, such as whether the problem
is unique to our patches or also exists in the official firmware.
Make sure that your title accurately describes the problem.

Wolf-DL4YHF, 2017-02-13:
Exactly that's the purpose.
"Leave the setting unchanged", if the current setting is NEITHER 5, 30, 
or 60 seconds
(but one of the settings that you can only select in Tytera's standard 
menu).
This item ("Unchanged") is there to indicate that the current setting 
different from the options further down the list.

It's difficult to find a concise description that fits on the tiny display.
When you select "unchanged", it does nothing (but leave the setting, 
whatever it is, unchanged).

Without the "unchanged" item, which of the other 3 (5, 30 or 60 seconds) 
should be marked when entering that menu screen, for example when the 
current setting is 10 or 15 seconds ?

Hope this explains why that menu items behaves as it does. "It's not a 
bug but a feature".
I am open for a better name, though.

All the best,
  Wolf .

IK8JHL, 2017-02-13:
Hmmm but its not like you say
I select  5 sec in original menu 
i select  60sec in md380tools menu and write " 60sec" 
after i select" unchanged" and menu write "60 sec" but I expect  to write the value of original menu "5 sec"

Wolf-DL4YHF, 2017-02-13:
Hi,

>
> Hmmm but its not like you say
> I select 5 sec in original menu
> i select 60sec in md380tools menu and write " 60sec"
> after i select" unchanged" and menu write "60 sec" but I expect to 
> write the value of original menu "5 sec"
No, because you already overwrote the 5 sec with the 60 sec in your 
second step.
If it would act as you described in your last line, we would need an 
additional memory location, to store the original value from the 
original ("Tytera") menu. Unnecessarily complex imo.
"unchanged" cannot UNDO the change that you performed by writing "60 
sec". There is only one memory location for this. Both menus (the one in 
the original menu and the one in the MD380Tools menu) both control the 
same memory location.

"unchanged" doesn't do anything when you press the 'confirm' button when 
selecting this item.
Again, it's purpose is an *indicator only*, to indicate that none of the 
other three items is currently in use.
An alternative text (label) would be "none of the settings listed below 
is currently in use" but that's too long for the display. So I called it 
"unchanged" because that's what it does when selecting/confirming it - 
it does no change anything, similar as pressing the "back"-button.

I hope this isn't too difficult to understand. "unchanged" or "other 
setting" or "don't change" or whatever you prefer.

travisgoodspeed, 2017-02-13:
Closing issue.  This seems to work as intended.

closed 2017-02-13 by travisgoodspeed