travisgoodspeed/md380tools #687: Latest Firmware backlight keep flashing after upload

VR2XKP, 2017-02-14:
I download the latest FW date 2-14-2017 from http://www.pd0zry.nl/md380-fw/ . I upload it to my MD-380 as usual but I found after upload and reboot, the back light keep flashing.

The purpose to upload to the latest FW is I found someone in Web told the back light can be setup to high /low now.

However, I resume it to older version (12-25-2016) , the MD-380 resume to normal but no such function.

travisgoodspeed, 2017-02-14:
How quickly does it flash, and what are the flasher settings in the MD380Tools menu?

Please use `md380-tool spiflashdump foo.bin` to export your SPI Flash and send me a copy of the file.

Cheers from New York,
--Travis

Wolf-DL4YHF, 2017-02-14:
I had a similar "flashing" effect (in the sense of a blinking backlight), caused by the firmware re-booting itself. Very difficult to say where exactly it crashes - due to a corrupted firmware image in the STM32's internal flash (from where the machine code is fetched) or due to corrupted / incompatible data in the SPI Flash (with the codeplug or our own configuration structure). 
So, without knowing if this is possible with the python utilities, it may also help to compare / verify the firmware in the CPU's internal flash. 

Another suspect (but this is highly speculative) is the way how we switch from "our" code into Tytera's own startup. Their original startup is designed to be run when the CPU comes out of reset, but in our special case, many on-chip peripherals are already initialised, and even some interrupts are already initialized, and the periodic SysTick interrupt is already running. If any of those interrupts fires at the wrong time (not really predictable), Tytera's own startup may crash or reboot the system (because their code activates the hardware watchdog, as can be seen in the disassembled listing). If the problem cannot be fixed by re-flashing or with a "clean" SPI flash, it may pay off to have a closer look at this. 

All the best,
 Wolf .

ejosterberg, 2017-02-14:
I would describe my experience as the screen is unlit or "dark/black" except for the tiny fraction of a second after a button press. I am only able to navigate the menu blindly until I can find the menu option to set the brightness level to something other than darkest setting and then acceptable back lighting is available. Can we default to some reasonable settings? Full brightness? A user can then choose other settings while being able to view the new menu structure? I banged around blindly poking buttons until discovering a setting that allowed me to view my screen.

pakomen, 2017-02-14:
I had similar problem with black screen, I changed code plug backlight time(s) from always to 5secs and the firmware upgrade worked fine after the change to the codeplug. I used this method on 2 x md380s and it worked on both.

Wolf-DL4YHF, 2017-02-15:
This may be related to the problem already mentioned some time ago - we 
may be hooking into the wrong function to initialize something.
IIRC in my case (D13.020-based firmware), the problem was related with 
the following setting:
"Radio Settings" .. "Intro Screen" : "Char String" was ok, "Picture" was 
not.
When not set to "Char String", one of 'our' functions doesn't seem to be 
called.
And indeed, the default backlight intensity should be "completely dark" 
during idle and "max brightness" when active, because that's the only 
way to avoid 'hum' in certain radios.

I will look into the 'hooking problem' myself as soon as time permits.

Cheers,
  Wolf.

Wolf-DL4YHF, 2017-02-15:
> via f299c1d

Hoping I'm not re-opening the issue, but there's one thing to consider:

The upper 4 bits were intended for the backlight when *active* (timer 
running).
The lower 4 bits were intended to be used when the activity timer timed out.
So, with 0x09 as default at that point, the display would be at max 
brightness when idle, and off after pressing any button. Maybe use 0x99 
to have the display at full brightness until configuring it.

I still think the real culprit is the (sometimes missing) call of the 
hooked init function.

travisgoodspeed, 2017-02-15:
Reopened the issue.  Let's close it when all is in order.

VR2XKP, 2017-02-15:
Hello all

I refer pakomen's information, set the backlight from always to 5s before upload the firmware, and the problem was solved.

And I resume the setting to always afterward and it is OK too.

Thank you very much for yours help.

Wolf-DL4YHF, 2017-02-15:
Ok, that's a workaround but not a solution. I'm working on it right now, 
maybe interrupts need to be disabled while reconfiguring some of the 
inner guts (RCC, GPIO, UART).
If nothing else helps, i will simply check in the periodic interrupt 
(SysTick_Handler) if everything concerning the backlight is properly set 
up, and if not (because some black box in the original firmware 
interfered), simply initialize it again - with all interrupts 
temporarily disabled. This isn't nice but appears logical...

More later,
    Wolf .

Wolf-DL4YHF, 2017-02-18:
Modified irq_handlers.c pull-request is pending ( #694 ).

"Waiting" until the original firmware turns on the backlight, instead of turning it on in the splash-hook-function, is the safer and thus preferred method. 

The backlight flashes up about a second after POWER-ON now (that's when the original Tytera firmware turns it on, and when it does, the internal voltage supply is ready for business).

Deliberately turning the backlight on EARLIER, in the first SysTick_Handler call, immediately crashed the CPU - guess the power supply needs some time to stabilize after the firmware (!) enabled an internal switching-mode regulator.  This radio is always good for a surprise !

KC5CQW, 2017-02-23:
Observation: mine flashes bright momentarily during USB TX to the radio. 
I believe this is the situation described in the original issue.

Wolf-DL4YHF, 2017-02-23:
Thanks for reporting. The reason is simple to explain, but not easily fixed:

Some of the USB handlers lock all interrupts for a long time. During 
this time, the backlight goes to maximum brightness because after a few 
milliseconds, the 'abused' serial port that drives the backlight, runs 
empty.
When the serial port's transmit buffer runs empty, there is no more 
pulse-width modulating pattern (104 Hz rectangular wave).
Without the PWM, the backlight (fortunately) falls back to maximum 
brightness.

This could only be fixed by avoiding those long 'critical sections' 
(with locked interrupts, or the system being 'very busy' from 
high-priority interrupts). Very tricky. Hope this is limited to the USB 
handler, which isn't in use during normal operation.

73, Wolf .

travisgoodspeed, 2017-02-23:
Closing this issue with the new merge.  The blinking during critical sections is regrettable, but unavoidable.

commit f299c1d9 (2017-02-15): The intensity default was 0x90, but it should have been 0x09.  Close #687.
closed 2017-02-23 by travisgoodspeed