travisgoodspeed/md380tools #688: Last update D13.020 today reset

PE2K, 2017-02-15:
D13.020 update today gives me a lot of reset in normale receiving. I never had that problem. Its after the blacklight update. On the MD380 from Tytera.

travisgoodspeed, 2017-02-15:
@PE2K Can you please describe exactly which settings are needed to reproduce the reset?

PE2K, 2017-02-15:
The Tytera reboots / reset itsself in a normal setting.

travisgoodspeed, 2017-02-15:
I really need a lot more detail than that to reproduce your problem on my radio.  If I can't reproduce your problem, I can't fix it.

travisgoodspeed, 2017-02-15:
I might have fixed your problem with my fix to #687.  Please rebuild from source and let me know if the problem persists.

PE2K, 2017-02-15:
I have update it. And testing it.

PE2K, 2017-02-16:
Its working fine. No resets any more. 73 PE2K

closed 2017-02-16 by travisgoodspeed