travisgoodspeed/md380tools #693: Backlight / button assignment Issue with  f082d02 2017-02-16

LU4CN, 2017-02-16:
Hi, side button assignment for backlight toggling on/off is not working after updating to f082d02 (2017-02-16). Best regards.

Wolf-DL4YHF, 2017-02-16:
To the developers of the side button control: This may be related to the 
entirely 'activity timer driven' backlight control in irq_handlers.c : 
SysTick_Handler. Just bit-banging GPIO C bit 6 to turn the backlight on 
/ off somewhere will not work. Is there already a global flag for this, 
i.e. "backlight manually turned on" ? Is it the same as '..backlight_on' 
in gfx.c which, when I looked at it some weeks ago, didn't seem to have 
any function at all ?

Cheers,
   Wolf .

LU4CN, 2017-02-16:
Actually, the issue is that backlight does not turn "off" when pressed. Thanks.

claywalker0929, 2017-02-17:
Thanks Wolf, your comment helped me know exactly where to look. Thanks LU4CN for reporting. Got a PR going through hopefully soon.

Wolf-DL4YHF, 2017-02-17:
Ok Clay, since there is another PR pending for irq_handlers.c, I will 
already merge your new flag there (guess there will be a merge collision).
If I understand your patch correctly, kb_backlight = 1 means "backlight 
shall be COMPLETELY OFF (not just dimmed), even if the activity timer is 
not expired yet " ?
At least that's what the condition
   "if( intensity == 0 || kb_backlight)"
would actually do.

The need for IRQ_Init() has been completely eliminated. But that's 
another subject.

Cheers,
   Wolf

claywalker0929, 2017-02-17:
Yeah, you got it. I closed my PR. You can take over and change the implementation to suit your style. I'm not too picky on this particular issue.

Wolf-DL4YHF, 2017-02-17:
Ok thanks Clay. I copied your modification into irq_handlers.c as-is. The only thing I wanted to change was to have kb_backlight declared as 'external' in keyb.h .. otherwise there may be linker errors when including it from multiple modules (which, at the moment, irq_handlers.c doesn't, but this may change very soon).

Is there a way to configure the backlight sidekey (etc) in the stock CPS, or does it only support config items that existed in the original firmware ? I'm not really familar with the Tytera / Retevis CPS...

claywalker0929, 2017-02-17:
I saw that. I was unaware of that issue until you commented on it and I did
some research. So thanks for that.

As for the CPS software, no you can't set it there, only in the menu. One
nice thing is CPS won't overwrite that value unless you actually select a
new value. I created a python script to be able to edit a codeplug and
included the side buttons as the first function. However I don't believe
this is useful anymore and in fact will probably delete it soon.

travisgoodspeed, 2017-02-25:
This this issue still a problem, or was it fixed with recent merges?

LU4CN, 2017-02-25:
Yes, it was solved. Thanks.

travisgoodspeed, 2017-02-25:
Excellent! Closing issue.

closed 2017-02-25 by travisgoodspeed