travisgoodspeed/md380tools #696: change SPI to 16M VHF version

autodiag, 2017-02-17:
have question for SPI eeprom 25qxx
i have VHF radio,inside 1M spi (25q80)..i am change this to 16M version (28q128 program to original data) but LCD white after power and CPS give error wrong model if go read or write.
Base FW check SPI ID or check memory type? Known when UHF have 16M flash and work..why dont work to VHF version?
I tried it cleans even played in a programmer and also through SPI read / write to MD tools
files are the same
When compared to the original 1M and 16M is the first new 1M same (to 0x100000)
CPU with normally communicates via MD TOOLS to eeprom..possible enroll for CPS format via md tools
radio but still has white LCD and communicating with CPS

danny1990, 2017-02-22:
Try flashing a stock firmware for your radio version IE 2.XXX or 13.XXXX to see if that resolved the problem. Make sure you flash the correct version or you may cause more damage.

Take a look here for the firmware and software:
http://www.va3xpr.net/programming-software-firmware/

autodiag, 2017-02-23:
sorry danny i am dont have any damage and dont have any problem to oriiginal 1Mb size eeprom
onli dispute what problem to echange 25Q80 to 25Q128....same protocol,same memory bank mapping but 25Q128 (16Mb) its problem to VHF version....but for UHF version use mean ,any problem...my ask is what diference

travisgoodspeed, 2017-02-23:
Auto, what is the JEDEC ID of your 25Q128?  Does it exactly match the JEDEC ID of the UHF model?

travisgoodspeed, 2017-02-25:
Closing.  Please move this to the mailing list or IRC.

closed 2017-02-25 by travisgoodspeed

autodiag, 2017-02-26:
hi

flash ID is C8 40 18

travisgoodspeed, 2017-02-26:
The firmware is probably looking for EF 40 18.  You will also need to move your calibration data over.

Not reopening.  Move this to the mailing list or IRC.

autodiag, 2017-02-28:
how cal data change and change EF 40 18 to C8 40 18 in fw?
sorry but where mailing or IRC..many thanks