travisgoodspeed/md380tools #700: no valid user list in stripped.csv

s-s-s, 2017-02-21:
The procedure make flashdb or make or updatedb flashdb or make updatedb_eur flashdb generates no valid user list for my county at least - DMR ID does not match the Callsign.
But at the same time, in the mmdvm code the list of users in the file  https://github.com/g4klx/MMDVMHost/blob/master/DMRIds.dat  is created correctly.
It  is from stripped.csv
2500001,R8AAK,Aleksey,,,,Russia
2500002,R2DFR,Mikhail,,,,Russia
2500003,R2DDM,Valery,,,,Russia
2500004,R2DDS,Vladimir,,,,Russia
2500005,R2AEW,Branislav,,,,Russia

It  is right data from DMRIds.dat
2500001,RU0LA					
2500002,R0AET,Vladimir				
2500003,R0ADW,	Alexandr				
2500004,R0AFJ,Gleb				
2500005,UB0AGO,Evgeniy

KD4Z, 2017-02-22:
The data for stripped.csv comes directly from dmr-marc.net, and is not altered or maintained by md380tools.    If that data is inaccurate, the individuals having incorrect data will need to send an email to idteam@dmr-marc.net and request a database change.   The scripting used by updatedb is operating as designed.

s-s-s, 2017-02-23:
It is possible to add a option and the right link http://registry.dstar.su/dmr/DMRIds.php to download stripped.cvs for HAMs of the BrandMeister network? Not dmr-marc.
Today we do it manually by command
` sudo wget http://registry.dstar.su/dmr/DMRIds.php`
After that edit stripped.cvs and flash db.

KD4Z, 2017-02-23:
Feature change requests should be posted in the Google support group:  https://groups.google.com/forum/#!forum/md380tools

This focus of this forum is to document reproduceable issues to be addressed by the md380tools dev team.
The format of the file obtained from the registry.dstar.su like is very different than the existing dmr-marc export, and seems to contain far less information.  Although it is possible to mechanically convert that file format for compatibility with the dmr-marc format,  it is doubtful that support for this alternate source will be added to the main md380tools project.

closed 2017-02-23 by travisgoodspeed