travisgoodspeed/md380tools #701: Sidekeys / activation of Morse code output for visually impaired hams ?

Wolf-DL4YHF, 2017-02-22:
I have a Morse code output with adjustable tone frequency and volume running now (independent from the analog pot). It uses the already existing, 'hooked' SysTick handler. Sounds great, but to turn this into a really useful extension (as requested some months ago in another thread), a lot of work is still required.

First of all (and this is where help from others would be much appreciated):
- how can an interrupt handler (which must not call API functions) find out the menu has been entered ?
- how can the same handler retrieve the text (string) of the currently menu item ?

With almost all side- and front-keys already occupied by Netmon or the freely programmable sidebuttons, how could a Morse audio output be activated "on request" ?
- use a fixed button (e.g. a short press of the "M" key below the PTT) ?
- or add it as a new function for the freely programmable side buttons ?

If anyone here likes to try out / test the Morse output, I can put it into the temporary branch that's been waiting to be merged since a couple of days (PR #694). I understand that the Netmon / Talker Alias feature is the 'main building site' at the moment, so I guess it will take some time until a really useful Morse-'narrator' is available.

Cheers,
  Wolf .

DL2MF, 2017-03-04:
This is really a nice extension, I had already a look into your fork. When TA has reached a sophisticated level for me, I'll try it out.

Best regards,
Mike

Wolf-DL4YHF, 2017-03-05:
There's "light at the end of the tunnel" - I found a non-intrusive way to find out the currently *focused* menu item myself. The global variable 'currently_selected_menu_entry' isn't just the "checked" (previously confirmed) item in a list of radio-buttons. It also changes nicely, immediately, whenever a cursor up / cursor down key is pressed in any menu (main menu, submenu, list of radiobuttons, etc).

Closing this issue myself because it's not a question anymore, to avoid possibly duplicate with #580 .

closed 2017-03-05 by Wolf-DL4YHF