travisgoodspeed/md380tools #702: Char String values "unknown"

KC5CQW, 2017-02-23:
Starting with 02182017 fw, the char. string user text is replaced by " unknown ". Also, startup tones are silent (I'm actually ok with this!) 
To confirm the issue, I reflashed the OEM fw S13.32 and a new blank CP. Then I reflashed back to the 02182017 release and my personal CP. Issue returned.

KC5CQW, 2017-02-23:
This looks to have corrected itself on my 390G after dropping it on the floor from an arm rest, (not kidding). I will get a flash dump and video if it returns... Perhaps this was just an isolated event.

closed 2017-02-23 by KC5CQW

travisgoodspeed, 2017-02-23:
Please use the mailing list for isolated tech support.  The issue tracker is only for reproducible issues.