travisgoodspeed/md380tools #703: Minimum DMR RX audio volume level

KC5CQW, 2017-02-23:
The DMR RX volume remains quite audible even at a zero setting. From the 0-9 scale, only 3 or 4 actual steps are observed. 0-3 seems to remain at a 3 (approximately).  On screen volume changes are indicated randomly with no user input. "Roger Beep" volume does not track with RX audio. Volume settings of higher value render the "RB" inaudible. (Perhaps this is a static value similar to the other radio tones and not yet a fully matured feature).
Analog RX audio seems to work normally.

KC5CQW, 2017-02-23:
FW: 02182017

travisgoodspeed, 2017-02-23:
Volume control is analog, implemented by a potentiometer rather than in software.  This is explained in issues #98, #124, and #184.

Closing this issue as a duplicate, but I've added the beep consistency request to #370.

closed 2017-02-23 by travisgoodspeed