travisgoodspeed/md380tools #704: Cleanup Superfluous Warnings

travisgoodspeed, 2017-02-23:
We have too many warnings in the source code, particularly when building for the 2.032 target.  We need to clean them up in order to make it easier to identify new bugs.  For example, pull request #694 was briefly failing unit tests for reasons that were obscured by all these warnings.

Tag commits that clean up warnings with this issue's number, and we'll close the issue when the bulk of the warnings are gone.

sijskes, 2017-02-23:
shall we convert the #warning directives and convert them into //TODO ? easy to find with smart-ide or grep.

travisgoodspeed, 2017-02-23:
Those are a sticky case.  Does GNU's preprocessor allow us to do a different warning type, one that would show in one case but not another?

claywalker0929, 2017-02-23:
Can we discuss dropping support for 2.032 firmware? Is there any good
reason for keeping it?

travisgoodspeed, 2017-02-23:
The only reason for keeping it is that it is the only branch that boots on all hardware, which is very handy when bootstrapping new models.  We can drop it if we get D13 booting on S13 hardware, or S13 booting on D13 hardware.

dschaafsma, 2017-04-25:
>On Thu, Feb 23, 2017, 7:12 AM Travis Goodspeed ***@***.***> wrote:
> Those are a sticky case. Does GNU's preprocessor allow us to do a different warning type, one that 
> would show in one case but not another?

FYI, gcc does allow suppression of "#warning" directives via compile flag "-Wno-cpp"
I verified this works in Cygwin (gcc 4.9.3) as well as Debian8.3-jessie (gcc 4.9.2)

From the gcc docs:
-Wno-cpp     (C, Objective-C, C++, Objective-C++ and Fortran only)
Suppress warning messages emitted by #warning directives.