travisgoodspeed/md380tools #705: Hang and restart in sidebutton

IK8JHL, 2017-02-23:
After  update 23-02-2017 
1) I have all sound disable , but sometime i listen beep when i touch key and some not
2)Whe i go in sidebutton  and i try to select Up/Down , the radio hang and restart

# PLEASE READ ME! #

When you click "Submit new issue" below, you are sending an email to
every developer of this project.  Our phones buzz, our computers beep,
and we are distracted from whatever we were doing.

Github Issues are not a discussion forum, and they are not a place to
share nifty ideas for new features or to ask for help.  Please discuss
those things on the mailing list, which is a forum that welcomes ideas
and discussion.

https://groups.google.com/forum/#!forum/md380tools

If you are certain that your bug is serious or a project developer has
agreed to implement a feature, then erase this text and replace it
with a clear explanation of exactly (1) what the problem is, (2) how
to reproduce the problem, and (3) what the expected behavior would be.
Please include as much detail as possible, such as whether the problem
is unique to our patches or also exists in the official firmware.
Make sure that your title accurately describes the problem.

travisgoodspeed, 2017-02-23:
Questo è troppo corto; non riesco a riprodurre. Si prega di utilizzare IRC o mailing list. Forse si scrive italiano?

Closing issue.

closed 2017-02-23 by travisgoodspeed

IK8JHL, 2017-02-23:
Sorry You have reason ,