travisgoodspeed/md380tools #706: Netmon not working properly

mi0hoz, 2017-02-23:
Only one entry shows up in Netmon.  It happened from the Feb 17th update.

travisgoodspeed, 2017-02-23:
Please provide a more complete explanation and instructions for reproducing the issue.  Photographs would also be handy.

mi0hoz, 2017-02-23:
Up until last Friday Netmon 4 - Lastheard showed up to 10 entries. Now it
only shows one entry.

mi0hoz, 2017-02-23:
MD380 Tools ver: 2017-02-23 CP Ver01.30
M. RogerBeep Disable
Boot Options Demo screen Disable
Date format DD/MM/YYYY
UsersCSV Enable
USB Logging Disable
Promiscuous Enable
Mic bargraph Enable
Experimental Disable
Side Buttons
         Top Pressed High/low Pwr
         Bottom Pressed   Scan On/Off
         Top Held Monitor
         Bottom Held Talkaround
Backlight
         Level Low  0  (Off)
         Level High  3 (Medium)
         Backlight Tmr  30 sec
CoPL Override  Callsign+Name
DevOnly!!    Enable

travisgoodspeed, 2017-02-23:
Building the current master branch from source and flashing a radio device with D13.020, I cannot reproduce this.  When I tune to 441MHz simplex and transmit three times from my Motorola, I see three entries of `KK4VCZ->99` in my Lastheard list.

Any other ideas for reproducing this problem?  I can't fix a bug that I can't see.

travisgoodspeed, 2017-02-24:
Can anyone reproduce this issue or provide a video/photos of the problem?

mi0hoz, 2017-02-24:
I am busy at the moment but will do a video for you tomorrow

UB1AAM, 2017-02-25:
[https://www.youtube.com/watch?v=V8l_RwUORX8](url)

mi0hoz, 2017-02-25:
https://www.youtube.com/watch?v=6HXvxvwcSkk

Wolf-DL4YHF, 2017-02-27:
Similar Netmon problem here. Looks like a multitasking isse, it seems to 
depend on the CPU load *and* from where the call to netmon is made 
(there may be several tasks from the real time OS involved).
Here is a very strange effect:
When activating netmon through a numeric key, sometimes the screen 
remains unchanged, sometimes completely black, and sometimes  a 
*partial* display of the netmon console appears.
When the screen appears to freeze, is it still possible to enter the 
menu, navigate in it, and leave it via the "back"-button (i know because 
the Morse-"narrator" already works to some degree, and correctly reports 
channel switches + menu entry/exit by watching various global variables).

In one or two (rare) ocassions, the display didn't return even after 
*shortly* turning the power off and on again.
It took a few seconds (!) with the power "Off" to get it alive again. 
Sounds like an 'accidentally misconfigured register' in the LCD 
controller, because the GPIO-registers are all actively re-configured by 
the system startup code.

I suspect there is 'interference' between the graphic output from netmon 
and some other tasks.
Beware: The LCD port is reconfigured between...
   - "driving the display" (LCD data bus configured as FSMC via the 
MODER- and AFCON registers)
   - "poll the keyboard" (LCD data bus configured as GPIO to *drive* the 
active rows/columns of the multiplexed keyboard)

Now what happens if we call certain graphic output functions (from a 
thread where gfx_.. must not be called) ?
'Garbage' accidentally written into the display controller !

It can be seen in the disassembled firmware that the switch between 'LCD 
output' and 'keyboard drive' doesn't take place for every little graphic 
output function (e.g. in D13.020 at 0x08051f22). In my disassembly 
listing I gave it the name "ConfigureLCDPort" because this is what I 
guess the function is supposed to do.
Not sure in which of the tasks it runs, but it's called from 
"biglist_pollsubsys_maybe" (I guess this will ring a bell for Travis). 
The other caller is a subroutine of the low-level keyboard polling 
function.
Again, I'm not sure what would happen if netmon() is called from the 
wrong task, and 'tries to' (!) output something while the LCD/keyboard 
interface hasn't been configured properly.

We should revise the hooks that call netmon() [there are a few], and 
find out if all their callers run in the same task that also writes to 
the display in the original firmware. But as usual, that's easier said 
than done.

For those familar with ARM assembly, the relevant function to switch 
from "keyboard polling" to "output to the LCD" is listed here: 
http://www.qsl.net/dl4yhf/RT3/listing.htm#ConfigureLCDPort

There are links to the callers above each function in the html, so you 
can use a web browser to easily navigate through the listing.

Another, possibly related item:
The software activates the hardware watchdog (see "FeedTheWatchdog" in 
the listing).
So far, I only saw it being called from the USB-IRQ-Handler (why??) and 
from "task_rtc_timer" (and endless task loop).
I don't think the time requred to paint the netmon / console screen is 
long enough to trigger the watchdog, but it's awkward to measure without 
a JTAG/SWD debugger (did someone already manage to connect one to the 
MD380) ?

Oh well, another post that grew longer than intended :)

Cheers,
   Wolf DL4YHF .

travisgoodspeed, 2017-02-27:
@Wolf-DL4YHF I've connected an SWD debugger.  Aside from the blue wires, you'll need to disable the STM32's Readout Device Protection (RDP) by using the jailbreak that I describe in PoC||GTFO 10:8.

travisgoodspeed, 2017-03-05:
I still don't understand how to reproduce this issue.  Could someone please write a proper description of the problem, and how to reproduce it?

Wolf-DL4YHF, 2017-03-05:
The problem is that it is not always there / behaves differently 
sometimes, and is affected by power-cycling the radio.

Step by step procedure to reproduce it here (on DMR+, TS1, TG262):
"Intro Screen" set to "Char String" (if that matters .. I didn't try)

Turn on, enter MD380Tools menu, DISABLE (!) DevOnly (reason gets clearer 
below)
Turn radio off, and on again.
Enter MD380Tools menu, ENABLE DevOnly.
Press BACK button three times -> back on the default screen
Press '4' : In many (but not all) cases, nothing happens - no console 
screen appears
Press MENU: In many (but not all) cases, nothing happens - it doesn't 
enter the menu
Turn channel knob : Console appears, netmon 4 / Lastheard appears
But none of the front button reacts.

Now the bizarre effect:
Turn radio off (DevOnly still enabled), wait, turn on again.
Wait for main screen to appear.
Press '4' : Almost always, Netmon 4 / Lastheard appears. No problem at all.
Press '7' : Netmon 4 quits, main screen appears again. No problem, too.
Press MENU: Menu appears, no problem again. Everything works as it should.

But now, without turning the radio off in between:
Enter DevOnly. It's still enabled, of course. "Confirm" without really 
changing anything
Press BACK until back in main screen. Ok so far, but..
Press '4' : Nothing happens ! Entering and leaving the menu (without 
changing anything) seems to bring up the problem again.
Turn channel knob: Kicks the Netmon 4 screen alive, but...
Press '7' : Nothing seems to (!) happen on the screen (still 
netmon/console there)
Turn channel knob: Similar as above, this seems to initiate a 'screen 
refresh',
and the main screen is back.
Press '4' : no reaction
Turn knob : ok, netmon
Press '7' : no reaction
Turn knob : main screen re-appears
And so on... sometimes this cycle can be repeated a dozen times.

I guess when new entries don't seem to (!) get displayed on the console, 
it's in fact the disturbed screen refresh which needs another "kick" 
from whatever is invoked when turning the channel knob.

DL2MF, 2017-03-06:
I did try to reproduce the issue following above procedure, but could not cause any error with my MD390/G trying this steps more than 10 times.

I'm using my radio most times about every day since mid december, the lastheard is being used several times a day. What I can confirm, that in very rare cases I did notice, that returning from a random netmon screen or entering menu is lagging. 

But the current gfx routines with hooks to the original firmware are hard to handle, like I had to notice, now working on the TA screen with refresh/update to the display information during rx.

mi0hoz, 2017-03-06:
I think that the dv4mini that I use could have an effect. I am not in range
of a repeater but I will be on Wednesday evening when I will test it again.

mi0hoz, 2017-03-09:
UPDATE.

I bought a new MD-380 and can reproduce the error. It only happens using a DV4Mini and using firmware after Feb 17th. If I use older firmware or work simplex or through a repeater, netmon works ok.

PD0DIB, 2017-03-09:
Can you please specify software versions and other details, so it is
possible to test? Things as "older" and "new" doesn't work for me. Hardware
model/version; firmware version/build, etc.. would be a great help...

mi0hoz, 2017-03-10:
> 285257903>,

TYT MD-380.   MD380 Tools version 07-03-2017.   DV4mini version 1.77
connected to a raspberry pi 3.

If I load firmware from Thursday 16th February the problem doesn't exist.

mi0hoz, 2017-03-11:
I just updated the dv4mini control software to 06.03.2017 and now everything works. I Guess this thread and issue is now closed.

closed 2017-03-11 by mi0hoz