travisgoodspeed/md380tools #709: MD-380/390 - with DV4Mini and BlueDV(BlueSpot)

ok2it, 2017-02-28:
Hi all
I have been asked to put "on paper" this:

it has been lately widely seens behaviour - I will explain:
1. DV4Mini or blueDV as HotSpot connected to DMR-BM Master
2. Receiving a voice stream from a station (station currently talking can enter DMR-BM by any means)
3. Short drop outs in RF  or internet

MD380 loses audio and displays no-sync!while green RX Led indicates there is transmission going on. Powering radio of, changing channel and returning back doesnt solve this and there is still silence.
When different radio is listening to same channel (I used Hytera AR685 and DM8000 mobile dmr rig)they both produce audio just fine.

this started to happen when TA has been implemented and when DV4Mini/BlueDV started to send TA data. 
Question: Could this be down to LC Frames being used as sync (they used to be empty FFFFFFFFF) and now, there is data in them? Maybe i am completely wrong, just a thought...

Anyway i managed to replicate this on both dv4mini and BlueDV with MD390G.

DV4Mini:   FW  1.75  and DV4MF2 client app.  MD-390G  (md380-tools from 25th FEB 2017).

I am happy to assist if there is anything I could provide...
Or perhaps workaround could be made by disabling TA for DV4Mini completely??

73

Jiri OK2IT

bradramsey, 2017-02-28:
Jiri and all,

Have you confirmed the lack of this behavior on a pre-TA revision (such as
commit 50CA4D28)?  I'm worried about the risk of inferring causation from
what may be a coincidence.  I sincerely hope that the sync detection is not
affected in the way that you theorize, as it would imply a poor design and
may be difficult to patch.

On that note, if there is a reference to the dsp functions and how to patch
them, I'd love to take a look.

Brad

ok2it, 2017-03-01:
Hey Brad,

Good to hear from you. I do have a limited knowledge and as a not English speaker it might have been misinterpreted.

I confirm that same behavior is seen on original S13.20. From TYT, MD380 Tools and also on original S13.09. This radio was shipped with. I therefore believe that this might be an issue more related to implementation of LC data within original firmware and possibly also coincidence with DV4Mini-BlueDV not sending these correctly? I do have confirmation that this also happens on old Motorola DP3601 radio (MOTOTRBO 1)which doesn’t cover DMR Tier2 as far as I am concerned, but AR685 decodes fine, even if turned off and on again.

Regardless I do stress out that I am not an expert and I came to conclusions with knowledge (limited) I have. I know that when TA is not sent, FF FF FF FF … data is being sent in LC space in regular occurrence which can be easily used as sync – thats why i thought this could be explanation…I am wondering if there is anything I could get you – logs, wireshark captures of DMR Streams (I do have MMDVM repeater, Vertex VXD-R70 Repeater, MMDVM HotSpot and DVMEGA MMDVM or BlueDV and DV4Mini also + MD390G , DP3601 Moto, HyteraAR685 and MYT DM8000. I am happy to assist if I can set up a lab and try out whatever could help you.

Again - I’m not an expert, just an enthusiast and ex moto-salesman..

Brad thanks a lot for looking into this.

PS . This behavior started on the day when BrandMeister Master has been upgraded to latest version (and when DV4Mini and BlueDV started Txing TA data).

travisgoodspeed, 2017-03-02:
Can someone with a protocol analyzer please confirm whether the MD380 or the access point is at fault?

IK8JHL, 2017-03-03:
But somthing its not clear !!! You say    "_MD380 loses audio and displays no-sync!while green RX Led indicates there is transmission going on. Powering radio of, changing channel and returning back doesnt solve this and there is still silence."_        But Dvmega and Dvmini not use sync because transmit in simplex in DMO mode , its very strange .... Im using Openspot without problem

ok2it, 2017-03-03:
Are you 100% sure DMO mode doesn’t use SYNC???
I would be surprised but you might be right…..even d-star uses sync and theres no difference in RMO and DMO.
I am not talking about DMR Handshake happening right after CSBK,but what I am saying is that MD390 displays NO SYNC, even when channel is in DMO MODE,

One common factor – DVMEGA/MMDVM hotspot doesn’t do this id MMDVMHost is used as driving SW.
BlueDV uses Home-brew repeater protocol
DV4Mini uses DV4Mini protocol.
(BlueDV apparentaly issued new version which addresses this… I didnt have time to test it yet).

IK8JHL, 2017-03-03:
I do not say that is not true, but I wanted to understand what was happening condition. But in  Netmon 1 show "nosync" ?

ok2it, 2017-03-03:
That is correct - nosync is the status I see, while green rx led is lid, and while HYT-AR685 decodes.

IK8JHL, 2017-03-03:
Yes i confirm also a friend of mine with DVMEGA on Android V1.0068 Firmware V3.14, and MD380 have nosync and no voice, but only with station from MMDVM,for example station from Hytera repeater or Openspot are received well

DL2MF, 2017-03-04:
You most often can't use different radios with the same frequency settings of DV4mini.
Check the freq offset for each of your radios and program the channel depending to this by CPS, then you can leave the dongle working on the predefined QRG.

ok2it, 2017-03-04:
Hi

That is only true when dv4mini listens for DMO signal. But here we talk about dv4mini and BlueDV TXing while DMR radios are in listen mode.

Jiří Čulák
Presales Engineer
FourteenIP EU - Czech republic
Tel +420 22888 0370
Mob +420 731 611577

closed 2017-05-14 by travisgoodspeed