travisgoodspeed/md380tools #712: Menu Button Stops Working With Experimental "ON"

jpbj, 2017-03-03:
Two issues:

1.  I updated the MD-380 firmware last week, and found that when I turn "Experimental" mode ON, then back out of the menus to the main screen, the "Menu" button no longer works to get back into the menu.  Turning off the radio and restarting it (of course, Experimental mode is toggled back to OFF), the menu button works as designed.  I updated the firmware again today to see if it went away, but the behavior persists.
2. I use a Char String on boot up. I like the bigger font, but I found that each line truncates after the 9th character. (It cuts off the last digit of my phone number!)

Caveat: I have some funky behavior with my radio that it will intermittently (but very frequently) shuts down when I key the transmitter, requiring a battery pull to wake it up. Don't think it's voltage since it affect 3 batteries, including a brand new one. It has happened on different firmware versions too. My radio may be possessed.

Thanks for your awesome efforts.

Old Vocoder
MD380Tools Ver: 2917-03-03
CP Ver: V01.32
Boot Up Screen: 
D13.020
6e2c1ef
2017-03-03

dragstor, 2017-03-07:
I can confirm this.

Tried it on MD380, MD390, RT3 and RT8. Exactly the same behavior as mentioned.

ok2it, 2017-03-13:
You can get to menu by pressing it and holding for a bit longer... ;-)

KG5IEN, 2017-05-09:
Issue 1 (menu button doesn't work when experimental mode is on) has been addressed by ee98da8.  Issue 2 (boot-up char string truncates after 9th character) was not investigated for or addressed by that commit.