travisgoodspeed/md380tools #715: White Screen MD380G

hb9trt, 2017-03-06:
Dear Developpers, 

After I read the initial information about writing here, I hope I make nothing wrong ;-)

I have to devices: 1 piece MD380 and 1 piece MD380G. 

There's no issue with the MD380, but since I compiled the new firmware on 2nd of march, I only get a whit screen on the MD380G. No initial welcome screen visible.  I use the command: 

make flash_S13

for the MD380G. Before I do a 

make clean

AS I told, The MD380 and the Retevis RT3 works fine. 

With the firmware version about 2 weeks ago all worked well. 

Also with a whitescreen MD380G I was able to read the settings with CPS software and also write. So I think anything runs, but not correct. 

Thank you for developping this software for the Tytera and Retevis DMR devices!!

vy73
Reto/hb9trt

hb9trt, 2017-03-07:
If I switch on/off several tims, maybe press also a key while switch on it starts correct. Some time white screen somtime it starts as it should. This efect only sinc newer versions. But it's only the whit screen. The radio works, I can send and receive and make a qso. So it must be anything with the initialisation of the screen. Maybe...

vy 73 Reto/hb9trt

PD0DIB, 2017-03-07:
I know this can occur (reported 2 times in our FB group, both same problem
and solution) but was a hardware feature... Flat cable from PCB to screen
did not make contact.
Disassembled the radio, disconnected flat cables (be careful), clean the
connectors and replaced them. Problem solved.

hb9trt, 2017-03-08:
Hello PD0DIB

thanks for the info. This make sense. I'll do it this evening. And so I think this is no real issue in software. I'll close it. 

vy73
Reto/hb9trt

closed 2017-03-08 by hb9trt

PD0DIB, 2017-03-08:
Look for the 'how-to" disassemble document in the filesection from the TYT
MD380 Tools Facebook page.
Good luck, hope you will fix it.
73 de PD0DIB

hb9trt, 2017-03-08:
Hi PD0DIB
I did the work this afternoon, because I disassembled an md380 before, it was no problem. Then I removed the flat cable between display and board and clean the connections. Until now it seems to work fine. Thanks for the hint. 

Reto, hb9trt