travisgoodspeed/md380tools #717: Menu Items Text Messed

dragstor, 2017-03-07:
As I mentioned in the #656 last comment, when testing out Serbian localization strings, wierd things happen.

I'd like you to take a brief look at the following "screenshots": https://goo.gl/photos/EuH7L3PEQmBd7PtX7

As you can see in the each photo's description, some strings are messed up, for example:
 - under Contacts, 3rd item - "Manual Dial". It should say "Rucno biraj", but instead it says "Sanual Dial". 

Another example:
 - "confirm" bottom-left button is translated as "Potvrdi", but it says "tonfirm".
and so on.

Like if it's being merged with some other string... Can't figure out which ones.

Anyone of you localizers experienced this?

dragstor, 2017-03-09:
After a brief examination, I figured that somehow strings mem locations sort of change places :) After running rabin2 upon default image, some strings reside on totally different location, compared to a localized ones. I'll provide examples soon.

travisgoodspeed, 2017-03-09:
The localizing program was tossed together on a domestic flight.  It certainly needs a cleanup or a rewrite.

Self assigning, but I haven't got much time this week, so feel free to try it yourself.

dragstor, 2017-03-09:
@travisgoodspeed Same situation here, ain't got that much of a time :(
I'll give it a shot whatsoever :) Tell me, did you gather strings from experimental bin, or from the original one?

travisgoodspeed, 2017-03-09:
Strings came crom the original, but we can add patching for the experimental strings next.

dragstor, 2017-03-09:
I think that won't be needed... 
Let me test some patches first locally :) 
I'll report back after those tests.
Thanks!

on3svb, 2017-08-28:
Same error (text messed up) also with French firmware  that I did compile. Fixed error (see merge #808 ) for file localise.c but same results.

73's de ON3SVB