travisgoodspeed/md380tools #718: Display Volume Indicator

KC5CQW, 2017-03-08:
FW: S13.20.20170307-180001-f45b541
MD-390G

Loud audio into microphone during RX (blow into mic w/o PTT) causes the onscreen volume change Indicator to display momentarily. 

I suspect the VOX trigger is actuating. (VOX is not currently active or used on my radio).

travisgoodspeed, 2017-03-08:
This is a known issue, described best in #98.  What's going on is that the volume level is read in analog from the potentiometer, and that this value drifts a bit.  When it drifts across the threshold, the firmware thinks that you are changing the volume knob and updates the screen to indicate that.

On #370, we have a number of feature requests to change this behavior, but it's not a very high priority of this project.

73 from New York,
--Travis

closed 2017-03-08 by travisgoodspeed

KC5CQW, 2017-03-08:
Sorry, I cannot find anything that relates in issue #98 to the microphone activating the digital portion of the radio when TX is not requested. 
I'm saying that my inert radio in RX is affecting a non-analog/audio, logic controlled system by way of strong audio input from the microphone only. No movement of the volume knob is performed at the time of the glitch. My concern is that other systems associated with the digital portion may also be affected.

travisgoodspeed, 2017-03-08:
The volume circuitry is analog, not digital.  It has all sorts of weird behaviors because of that, which mostly involve the volume graph appearing at weird times.  This is not VOX, and your radio is not transmitting.