travisgoodspeed/md380tools #721: First MD380 blank screen

Giorgiofox, 2017-03-15:
Hi All, 
my MD380 first series, goes in blank screen after few second before boot. Sometimes minutes, sometime s seconds. It continue to work but the screen is all white. 
I've tried make flash, flash_D02, flash_D13. 

This not happen with original D02. 
What I can try?
thanks in advance
Giorgio IZ2XBZ

PD0DIB, 2017-03-16:
Think this type of questions do not belong on GitHub. GitHub is for
development, not for support.. Better place is to ask for help in the
supporting Yahoo group..
Hope you can solve it.

Giorgiofox, 2017-03-16:
This is an issue relative to md380tools, not a simple question. With older version of md380tool this not happened. As I know all the first md380 with new patched firmware has this problem

travisgoodspeed, 2017-03-16:
Georgio, can you give us a bit more information?  It's hard to reproduce your problem from this description, and we can't fix what we can't reproduce.

Giorgiofox, 2017-03-18:
Sorry for delay Travis. 
I have no many more details about this issue: this is a very first series md380 from Tytera, it work perfectly with original D02 firmware, no screen problem, no other problems. 

If I try to install md380tools with "make flash" after few minutes of normal operating the screen goes blank. If I turn off and on, radio work for few seconds (sometimes second, sometimes minutes) and the screen goes blank again. 

I have tried with all makefile options, but without success, except with original D02 firmware: so I think is a patched firmware related issue, but I can't solve. 

Maybe is possible try to erase all md380 memory before flash new firmware?

KD4Z, 2017-03-18:
Travis,
This may be a boundary condition I have seen many times when flashing a non-md380tools radio (virgin) to the current builds after the backlight dimming merge.   Every time I bring one forward, either the backlight stays on, or goes of rapidly.  The workaround I always do is make sure to write values for each of the three items under the new Backlight menu, immediately after flashing the radio.  I just change each dimming level up/down by one, and confirm.  I also set the time out to 5 secs usually.  (This may actually be the only one needing a change for the workaround)   After doing this, the settings can be adjusted to anything the user wants and it stays working.    I think the missing values read from settings,  when the radio is virgin, causes the issue.  Once set, it never happens again.   Giorgio, you might try this procedure after re-flashing the latest firmware.   Warren, KD4Z

Giorgiofox, 2017-03-18:
I can try to set this parameters after flashing, but I'm not sure this can be useful with blank screen issue, but maybe with backlight screen issue. 
My radio is not virigin, I've flashed it really many many times beginning about 1 year ago. 
I will try and I will report here if something changes

ispkorte, 2017-03-19:
if you use flash , you get D13 but u need D03 , so read manual to flash d03 version

Giorgiofox, 2017-03-19:
Why the D03 has removed from makefile?
How I can patch it?

travisgoodspeed, 2017-04-27:
@Giorgiofox Are you still seeing this problem?  I think #736 will fix the trouble permanently.

closed 2017-05-13 by travisgoodspeed