travisgoodspeed/md380tools #722: Change Channel power down the GPS on MD380G .

virtualrobotix, 2017-03-16:
Dear Developer,
i'm testing the MD380G Radio and found some iusse on it .
Some i solved changing the GPS antenna , but other is not simple to solve because my opinion is that on GPS there isn't the backup battery . 
So the other iusse in the software is that when you change the channel and going on channel not properly configurated for gps the gps is power down and when you switch to other channel with gps the gps restart in cold condition so need a lot of time to retake a correct gps lock. 
I try to check in the firmware where is manage the switch on and switch off of gps but i don't found the part of code where the power of gps is controlled ... 
So i love coding in .c if you can help to understand more about the code architecture i can produce and test a patch and commit it to the git . 
Best
Roberto Navoni

travisgoodspeed, 2017-03-16:
Howdy Roberto,

Our C code consists only of patches to Tytera's closed source binaries, and we haven't reverse engineered much of the GPS yet.

Do you know which pin from the STM32 controls the GPS power?  If so, we can search the stock firmware for functions that use that pin, and patch them to change the behavior.

-Travis

virtualrobotix, 2017-03-16:
I'll check during the  week what's the pin ... so i thought that the .c code wasn't fully complete because i don't found a lot of code :) 
Great Job guys 🥇

ok2it, 2017-05-23:
Hi Roberto

have you found what PIN controls the GPS power state?
I own TYT MD-390G (essentially same to yours but cover).

Thanks a lot
Jiri

eb4fbz, 2017-06-28:
GPS power is controlled with PC12 (Pin 80).

There is a NPN + PNP transistor pair switching the GPS power supply, so a high logic level should power up the GPS module. PCB photo later.