travisgoodspeed/md380tools #723: Backlight is always on

wemaus, 2017-03-17:
# PLEASE READ ME! #

When you click "Submit new issue" below, you are sending an email to
every developer of this project.  Our phones buzz, our computers beep,
and we are distracted from whatever we were doing.

Github Issues are not a discussion forum, and they are not a place to
share nifty ideas for new features or to ask for help.  Please discuss
those things on the mailing list, which is a forum that welcomes ideas
and discussion.

https://groups.google.com/forum/#!forum/md380tools

If you are certain that your bug is serious or a project developer has
agreed to implement a feature, then erase this text and replace it
with a clear explanation of exactly (1) what the problem is, (2) how
to reproduce the problem, and (3) what the expected behavior would be.
Please include as much detail as possible, such as whether the problem
is unique to our patches or also exists in the official firmware.
Make sure that your title accurately describes the problem.

Wolf-DL4YHF, 2017-03-25:
What do the other items in the backlight menu show (besides the backlight timer), i.e. "intensity low" + "intensity high" ? In all other cases I heard of, setting it once (in the menu) cured the issue.
Having both ("backlight intensity low" + "backight intensity high") at MAX brightness was a kludge added some weeks ago, because on *some* radios the backlight remained off, even when pressing a button (I could never reproduce that problem myself on an RT3 with firmware based on D13.020).

wemaus, 2017-03-28:
Thank you for your response. Now I have figured out that there is another Backlight (12) function in menu MD380Tools which I have not seen before. I have activated this and now it's working fine.
Is this function a replacement for the Backlight in "Utilities/Radio-Settings/Backlight (4)"?

Wolf-DL4YHF, 2017-03-28:
Indeed the backlight settings in the MD380Tools menu first offered some 
new (iirc longer) timeout values than in Tytera's own menu 
(utilities/radio-settings/backlight).
The different intensities were added much later, and seemed to cause 
problems on some radios, which is the reason why the intensity during 
'activity' and 'standby' were both set to maximum per default. This is 
just a kludge, and I don't understand why it was necessary on some 
radios (the problem never occurred here, on two different RT3's).

But better to have a display backlight that is initially always on 
(until configured) than one that is always off.
This will be fixed in a future version (when the reason is fully 
understood), but at the moment I'm too busy trying to understand another 
part of the firmware (related with menus, and how to retrieve the 
strings for output in Morse code).

Cheers, Wolf .

KD4Z, 2017-03-28:
Wolf,  I believe it's just the boundary condition of a radio not having any value for the timeout setting, when it FIRST gets firmware past the version having the dimming feature.  I see it everytime I update a virgin radio, or one that has tools firmware from last year only.   I've lost count of how many, but they always manifest the glitch after being flashed to dimming capable firmware.    I go in and change the timeout to something and the backlight timing works again.  A config reset fixes it as well, so whatever defaults it writes fixes the "null" being read right after a fresh upgrade.  Your radio has the values so never manifest the error.  
Warren

Wolf-DL4YHF, 2017-03-28:
Hi Warren,
Yes, very likely that's one of the causes. Maybe we should add a test 
feature to erase everything in 'our' config structue, to turn a radio 
back into the 'factory default state'. Same for the users database 
(users.csv) - sometimes it would be handy to find out how a new 
experimental firmware would behave on a brand-new radio.
73, Wolf.

travisgoodspeed, 2017-04-27:
I created #736 to add a sanity check to the menu structure to generically fix the problem.  It could be a nice quick pull request.

Closing this issue.

closed 2017-04-27 by travisgoodspeed