travisgoodspeed/md380tools #727: updatedb stopped working (noticed 2017-03-31)

dysinger, 2017-03-31:
I see HTML in the stripped.csv

```
user@oryx ~/s/g/t/md380tools> git clean -fdx ; git reset --hard HEAD
user@oryx ~/s/g/t/md380tools> make updatedb
user@oryx ~/s/g/t/md380tools> grep \<div db/stripped.csv 
,,,,,,lass="arrows frt_arr"></div>
,,,,,,lass="arrows lst_arr"></div>
,,,,,,lass="bottom_bg clearfix">
,,,,,,lass="bottom_rs clearfix">
,,,,,,lass="container clearfix">
,,,,,,lass="content clearfix">
,,,,,,lass="footer-nav">                
,,,,,,lass="header clearfix">
,,,,,,lass="kwd_bloack clearfix" id="main_rs"></div>
,,,,,,lass="leftblk">
<div class="sale-msg"><div style=" background: #001834; text-align:center; font-family:arial;">  <iframe id="custom-header-iframe" src="http://dm.ecommerce.com/error_header.html?domainname=d,,,,,,c.net" width="960px" height="60" scrolling="no" frameborder="0" style="height: 60px;"><p>Your browser does not support iframes. </p>;</iframe></div></div>
,,,,,,lass="searchbox clearfix" id="searchbox"></div>
,,,,,,d="main-wrap" class="clearfix">
user@oryx ~/s/g/t/md380tools> 
```

dysinger, 2017-03-31:
It is the dmr-marc data that does it (removing it as a dependencie target from the makefile works)

dysinger, 2017-03-31:
http://www.dmr-marc.net is f***ed.  Good job, everyone ;)

dysinger, 2017-03-31:
```
user@oryx ~/s/g/t/md380tools> whois dmr-marc.net | grep -i updated
   Updated Date: 31-mar-2017
Updated Date: 2017-03-31T13:51:18Z
```

dysinger, 2017-03-31:
emailing him

```
user@oryx ~/s/g/t/md380tools> whois dmr-marc.net | grep -i email
Registrar Abuse Contact Email: domainabuse@tucows.com
Registrant Email: paulmetzger@mail.com
Admin Email: paulmetzger@mail.com
Tech Email: paulmetzger@mail.com
user@oryx ~/s/g/t/md380tools> 
```

dysinger, 2017-03-31:
[image]

holgerschurig, 2017-04-01:
This night the brandmeister network got DDoSed, and also around 8am (german time) dmr-marc.net was not reachable. Sometimes MARC homepage itself was there, sometimes not. But in no case was their database accessible.

There is little that md380-tools can do here, except maybe better error checking :-/

dysinger, 2017-04-02:
I never got an answer from paul (above) but it looks like the site is back online today

closed 2017-04-02 by dysinger