travisgoodspeed/md380tools #730: Encryption settings at menu radiostation

gerronim, 2017-04-15:
Hello,

can i have a question about firmware for TYT radiostations MD380? I use firmware this version firmware-20170218-000001-221fcec3.bin and i am very satisfied. Do you have intention add to firmware function in menu for change enabled and disabled encryption on DMR channel? And change number of key basic and enhanced code? Add functions to menu of MD-380 radio. This would be great and the best of firmware for use.
Really. Tnx.

travisgoodspeed, 2017-04-15:
We do not currently plan to add such a feature.

73 from New York,
--Travis

closed 2017-04-15 by travisgoodspeed

gerronim, 2017-05-07:
> Od: Travis Goodspeed <notifications@github.com>
> Komu: travisgoodspeed/md380tools <md380tools@noreply.github.com>
>
> We do not currently plan to add such a feature.
>
> 73 from New York,
> --Travis

Thank you for response.
This funktion would be good, because programming radiostation would be without necessary PC.

You doing good work!

Best regards
73! Radim OK2XMW