travisgoodspeed/md380tools #732: Possible bug in netmon.c / call of print_time_hook() with wrong data type

Wolf-DL4YHF, 2017-04-17:
In the modified netmon.c, netmon4_update() calls print_time_hook(log) with a single (simple) character passed as function argument.

But in the modified gfx.c, print_time_hook() is declared as
void print_time_hook(char *log)

(Note that "log" is a char-POINTER here, not a char).

Suggestion:
Replace void print_time_hook(char *log)
   by      void print_time_hook(char log)

and put a proper function prototype where it belongs (e.g. gfx.h) .

I *guess* this was Mike's intended use of "log" - being a char, not a char * or string. 

Cheers,
 Wolf .

DL2MF, 2017-04-18:
@Wolf-DL4YHF: The parameter of print_time_hook is used to select, to which of the netmon logbuffers/screens the current entry will be written/added. I know a minor bug if you are already on one of the screens, when a call arrives. The timestamp is missing in this case, but that was not on my highest priority list before the latest merge.
I'll spend a look on the declarations when I have returned from vacation, this also will inhibit me to provide a compiled binary for testing purposes beside your dev environment to you before next weekend.

Wolf-DL4YHF, 2017-04-18:
Ok, I think we're not in a hurry because the pointer (which should be a simple character) isn't referenced as such. No risk of crashing with an access violation, Data Abort (Cortex exception) or similar.

In the meantime I brought down the number of warnings in my branch to exactly zero (the one with Morse output, and a new alternative 'flat' menu + much faster LCD driver), but it will take some time before pulling a merge-request for it.

DL2MF, 2017-04-19:
Hi Wolf-DL4YHF,

today I did want to try your latest build from your fork, but the:

make flash_S13

terminates with error:
`arm-none-eabi-gcc -std=gnu99 -g -O2 -Wall -Tapplet_s13.020.ld -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fshort-wchar -Wa,-a=narrator.lst -DFW_S13_020	 -Iinc -Ilib -Ilib/inc -I. -Ilib/inc/core -Ilib/inc/peripherals -D "MD380_s13_020"     -c -o narrator.o src/narrator.c
src/narrator.c: In function 'report_menu_item':
src/narrator.c:411:41: error: 'selected_contact_name_wstring' undeclared (first use in this function)
               MorseGen_AppendWideString(selected_contact_name_wstring );
                                         ^
src/narrator.c:411:41: note: each undeclared identifier is reported only once for each function it appears in
<eingebaut>: die Regel für Ziel „narrator.o“ scheiterte
make[1]: *** [narrator.o] Fehler 1
make[1]: Verzeichnis „/home/mg/md380tools/applet“ wird verlassen
Makefile:32: die Regel für Ziel „image_S13“ scheiterte
make: *** [image_S13] Fehler 2
`
Because I'm currently on vacation, I didn't spend any further time on debugging, just only wanted to give you a notice on that. The D13 make process is running fine, but not suitable for my radio.

DL2MF, 2017-04-19:
Because I can't open an issue or PR on your fork here my additions:

Please add in line #21 to narator.h:
`#if defined(FW_D13_020) || defined(FW_S13_020)`

Please add symbols in line #114-116: 
`msg_sms_hdr_prep                         = 0x2001cb54 ;
selected_contact_name_wstring            = 0x2001cb58 ;
msg_sms_bdy_prep                         = 0x2001ccbc ;`

Wolf-DL4YHF, 2017-04-19:
Hi Mike,

Am 19.04.2017 um 16:46 schrieb Mike:
> |msg_sms_hdr_prep = 0x2001cb54 ; selected_contact_name_wstring = 
> 0x2001cb58 ; msg_sms_bdy_prep = 0x2001ccbc ;|

Ok, thanks, I inserted these in symbols_s13.020 now.
Do you, by chance, also know the address of the current menu title ? In 
D13.020, it's a C-string (I assigned the name menu_title_cstring to it) 
at 0x2001e3c0. The address will be similar but not the same in S13.020 
.  At the moment all I have is an RT3 w/o GPS, so I cannot find the 
address myself. By inspecting the SRAM via USB while zapping through 
Tytera's menu it's easy to find.

Wolf-DL4YHF, 2017-04-26:
The pending PR #734 (with Morse output and the 'alternative menu') contains the described fix now.

Function prototype changed to
void print_time_hook(const char log)

('const' isn't really necessary, but doesn't do harm, and it was already suggested or even contained in another PR)

Wolf-DL4YHF, 2017-05-10:
Fixed by the recent two merges.

closed 2017-05-10 by Wolf-DL4YHF