travisgoodspeed/md380tools #740: Error in compiling.

IZ4FXC, 2017-05-06:
Hi folks!
I've reconfigured a new PC with Win7, reloaded all the necessary stuff and so no nu when I try to compile it seems that some files are missing. Iv'e tryied to re-clone everything but with no success.
Any advice?
Thanks in advance for your attention, Greg. 
[image]

KG5IEN, 2017-05-07:
If I'm interpreting the output (and Italian) correctly, I believe it is having trouble finding arm-none-eabi-gcc, which is part of the arm toolchain.  

Did you follow the [instructions here](https://github.com/travisgoodspeed/md380tools/blob/master/docs/gnuwin32.md)?  This snippet in particular seems relevant:

> Download and install gcc-arm-none-eabi from: https://launchpad.net/gcc-arm-embedded/4.8/4.8-2014-q1-update At the end of the install, tick the 'add path to environment variable' box before clicking finish.
>
> Reboot your PC

IZ4FXC, 2017-05-07:
Thank you for coming back to my question, yes I did and verifed the correct path variable settings too, all seems to be right.

Path: (Omissis...) ;C:\Progra~1\GnuWin32\bin;C:\Python27;C:\Program Files\Git\cmd

----Messaggio originale----

Data: 07/05/2017 17.48

A: "travisgoodspeed/md380tools"<md380tools@noreply.github.com>

Cc: "IZ4FXC"<demichele_g@libero.it>, "Author"<author@noreply.github.com>

KG5IEN, 2017-05-07:
I don't see the location for the arm toolchain in the path you posted.  On my machine (Windows 10 64-bit), C:\Program Files (x86)\GNU Tools ARM Embedded\4.8 2014q1\bin is the directory that is in my path that contains arm-none-eabi-gcc.exe

Is that directory (or something similar to it) in your path?  Can you find where the gcc-arm-embedded installer installed to and verify that arm-none-eabi-gcc.exe is in the bin subdirectory?

IZ4FXC, 2017-05-07:
Yes it is (in the omitted, sorry) but a reinstallation of GNU has solved the problem, maybe something didn't work in the first installation.
Thank you very much for having pointed my attention in the correct direction.
Hope to catch you in the frequency!
IZ4FXC, Greg.

----Messaggio originale----

Data: 07/05/2017 19.19

closed 2017-05-07 by travisgoodspeed