travisgoodspeed/md380tools #742: Help wanted for the PR #741 : Morse output & new 'Zone' list needs testing on GPS radios

Wolf-DL4YHF, 2017-05-08:
Greetings all,
After some struggle resolving all merge conflicts in the pending PR (now #741, since I deleted #743 for various reasons) is now ready to be pulled into the main repository. 
Since I don't have a GPS radio to test the new features, I cannot find out a few codeplug-related addresses within the SPI flash myself. I'll try to explain as short as possible:
- the new Morse output hooks into the original 'Tytera' menu. This works, but...
- some new features like entering a new talkgroup on the fly, or switch to a different ZONE (in the original menu) is extremely unpractical for visually impaired users
- because of that, I added a completely new menu, opened with the *RED* button (which has no function when on the main screen).
That new menu can list all zones from the codeplug, and select a new zone easily. But it completely bypasses the original firmware functions. Instead, it reads the list of zones direcly from the SPI-Flash (address range of the codeplug data). Works like a charm on an MD380 (including autorepeat and a fast scrolling list), and can properly announce everything in Morse code .

If you can afford the time, please check applet/src/codeplug.h (download from https://github.com/Wolf-DL4YHF/md380tools). There are new definitions for address-offsets within the SPI-flash, most related with codeplug data, e.g.: 

#define CODEPLUG_SPIFLASH_ADDR_ZONE_LIST  0x000149E0

Also, there are a few addresses for variables in RAM, where the original firmware keeps a copy of the current zone (for unknown reasons, there are TWO different places in RAM where Tytera stores the name of the current zone).
At the moment, the RAM addresses are only defined when compling for firmware D13.020.
I *guess* the codeplug addresses are the same for GPS radios because they use the same CPS.
So we need to check or confirm (before merging)....
 - if GPS radios store the 'Zone list' at the same address in SPI-flash,
 - where GPS radios store the ZONE NAME in RAM (zone_name[] and zone_name_2[] )
 - where GPS radios store the current ZONE NUMBER (this is an array index, 1 = first zone, etc)

Without knowing the RAM addresses, the zone list *display* should work also in GPS radios, but the possibility to select a different zone in the 'whitesticker-friendly' menu will not work.

Apologies for posting this as an 'issue' and letting your phones ring, but posting these questions in the md380tools user's group (on google) is unlikely to get an answer.

Also, if you have a GPS radio : Can you check if the Morse output volume can successfully be controlled via volume pot (despite the audio signal bypassing the pot) ? 

The Morse output can be turned on through the classic MD380tools menu, or (after pressing the red "back" button) in the alternative menu: Scroll down to 'Test/Setup', press enter (green key), scroll down to 'Morse output', and select mode 'Verbose'. The rest should be self-explaining, all configuration parameters are directly visible in the new menu.

Cheers,
  Wolf DL4YHF .

travisgoodspeed, 2017-05-09:
I confirm that this seems to work on GPS radios as merged to the master branch.  Volume does vary by the pot, but might be a bit louder than the channel audio.

Wolf-DL4YHF, 2017-05-09:
Yep... that's tricky to adjust, because the relation between analog pot 
reading and pulse width modulation (duty cycle) is very non-linear. 
Also, I had to add heavy low-pass filtering because the pot reading 
itself was extremely noisy. Maybe the lowpass is a bit *too* heavy, it 
takes one or two seconds to settle.

Wolf-DL4YHF, 2017-05-09:
Since others also got this to work on their GPS radios, only the question about the address of the Zone data in the SPI flash remains. But that's not the main point of this issue, so let's close it.

closed 2017-05-09 by Wolf-DL4YHF