travisgoodspeed/md380tools #747: No contact list menu

hsojhcoom, 2017-05-13:
Radio: TYT MD-380
OS: Linux Mint/ Win 10 with VM tools
Issues: Have flashed the firmware with both tools listed above. Both times it has given no menu option for contacts nor in the MDToolz Menu no option for contact list or options either. Tried first on the Win 10 with the VM tools. Ran GLV, flashed and updated db before flash db. 

Linux installed via direction on Github. No issues install, firmware install no issues (minus no contacts) Updated db, flashed db, no contacts still. Did a clean flash and flashed db with not luck

KD4Z, 2017-05-13:
For support of the md380tools, you should post in the Google group here: https://groups.google.com/forum/#!forum/md380tools

closed 2017-05-13 by travisgoodspeed

hsojhcoom, 2017-05-13:
There is no "User DB" Selection under the MD380tools menu......

Wolf-DL4YHF, 2017-05-13:
..... because it has been moved into submenu "Show Calls" :o)

I think we can close this.....