travisgoodspeed/md380tools #750: Website Feature Photos

travisgoodspeed, 2017-05-13:
Close this issue when we have nice photos for each visible feature on the website.  Updates to feature text should also be tagged with this issue.

To generate the website locally, install hugo 20.0 from http://gohugo.io/ and then run `make server` in `md380tools/md380org/`.  The master branch of the site gets pushed to https://md380.org/ every quarter-hour, but pages marked as drafts are left off of the public website.

Wolf-DL4YHF, 2017-05-14:
Is it possible to embed simple HTML tags in MD ?
If so, we could copy some of the "screens" (which are in fact plain old 
HTML table elements) from here, instead of using real screenshots 
(graphic images) :
http://www.qsl.net/dl4yhf/RT3/index.html#morse_output

(the same applies to the new alternative menu on which I'm currently 
working...)

Cheers,
   Wolf .

travisgoodspeed, 2017-05-14:
@Wolf-DL4YHF I pushed an example with a bad photograph of the call log feature to http://md380.org/features/calllog/

Images can be added as either Markdown or HTML, but in Markdown the dimensions can be auto-calculated.  Documentation for our theme's image types available at http://docdock.netlify.com/shortcode/image/

Please feel free to add whatever pages, text, and images you feel are needed.

Wolf-DL4YHF, 2017-05-15:
Greetings all,

I know this is crazy and possibly not worth the effort, but...
how about reading the LCD framebuffer via USB, and making graphic 
screenshots with a slightly modified md380tools python script ?
The principle used in applet/src/lcd_driver.c to move a rectangular area 
block of pixels from the CPU into the LCD controller (without having to 
send the coordinate for each pixel) also works the other way around, 
i.e. from LCD controller to the CPU.
The only pitfall would be that in the USB interrupt handler, we cannot 
safely assume that the FSMC (external "memory" interface) is really 
configured to access the LCD instead of polling the keyboard.

This would allow taking optimal "photos" (well, screenshots) for any 
kind of documentation.

Just food for thought..
73,  Wolf

travisgoodspeed, 2017-05-15:
Excellent idea!  Spun up as issue #754.