travisgoodspeed/md380tools #752: MD380 starting on zone number 4 after powering up

bartekpapierski, 2017-05-14:
Hi,

After I upgraded to the newest md380tools version with new menu under red button, my MD380 always starts with the same zone.
When I'm for example on zone number 2 and I power down the radio then power it up again, it always starts on zone 4.
I have about 15 zones configured and my 4th zone contains only analog simplex channels.

Bartek SO7BIT

Wolf-DL4YHF, 2017-05-14:
I guess you selected a new zone through the list in the new alternative, 
aka "red button menu". Indeed selecting a different zone THERE is not 
written back to the SPI-flash.
To permanently switch to a new zone, for the moment, select it in the 
original ("Tytera") Menu, and press Confirm to save your setting 
permanently.
In a future release the new menu (with the fast scrolling list + 
autorepeat keys) will also write back the new zone to the SPI flash.

Cheers,
   Wolf DL4YHF .

bartekpapierski, 2017-05-14:
Hi Wolf,

Yes, you are right, sorry for bothering. Closing the issue.

closed 2017-05-14 by bartekpapierski