travisgoodspeed/md380tools #754: Screenshots over USB

travisgoodspeed, 2017-05-15:
For documentation and for the website, it would be handy to take screenshots over USB.  Implementation suggestions by @Wolf-DL4YHF follow.

```
I know this is crazy and possibly not worth the effort, but...
how about reading the LCD framebuffer via USB, and making graphic
screenshots with a slightly modified md380tools python script ?
The principle used in applet/src/lcd_driver.c to move a rectangular area
block of pixels from the CPU into the LCD controller (without having to
send the coordinate for each pixel) also works the other way around,
i.e. from LCD controller to the CPU.
The only pitfall would be that in the USB interrupt handler, we cannot
safely assume that the FSMC (external "memory" interface) is really
configured to access the LCD instead of polling the keyboard.

This would allow taking optimal "photos" (well, screenshots) for any
kind of documentation.
```

Wolf-DL4YHF, 2017-05-25:
I'm working on this.. there is something strange with the FSMC timing registers (set by the original firmware) that violate the spec of the ST7735 (LCD controller). If the chip really is an ST7735, it's much faster on WRITE than on READ. So the read-accesses must be slowed down to at least 150 ns long low-active READ pulses, which is tricky with the CPU's 168 MHz "HCLK".
The recent PR #768 (read mfg string before using USB extensions) already helped a lot with the USB handler: the commands to read tiles from the framebuffer now get through. 
So even though the framebuffer only reads "0x00 0x00 0x00..." (at the moment), it's a step forward.

Wolf-DL4YHF, 2017-05-28:
Convincing the LCD controller to "spit out" pixels from the framebuffer was quite an effort but it working now (after reprogramming the FSMC unit, because the original timing parameters were not suitable).

Any Python experts or volunteers here to help with converting the raw pixels read from USB into a widely used image file format (BMP, or even better PNG, if that's not overly complex in Python) ?

A modfied md380_tool.py can already read out a block of 160 * 128 pixels, * 2 bytes per pixel, and store it in a binary file but that's nothing that a normal image file converter can read. So I think we need to do that in Python.

ea, 2017-05-28:
```
def screenshot(dfu, filename="screenshot.png"):
    """Reads the LCD framebuffer"""
    with open(filename, 'wb') as f:
      rows = []
      for y in xrange(0,128):  
        buf = dfu.read_framebuf_line(y)
        row = []
        for i in range(0,len(buf)-1,2):
            uint16 = struct.unpack("H",buf[i:i+2])[0]
            b = (uint16 & 0xf800) >> 11
            g = (uint16 & 0x07e0) >> 5
            r = (uint16 & 0x001f) 
            pixel = (r,g,b)
            row.append(pixel)
        rows.append(row)
      import png
      image = png.from_array(rows,"RGB",info=dict(height=128))
      print(image)
      image.save(filename)
      f.close()
```
Something like this aught to do it. 

Not sending a proper pull request since I can't get mine to spit out framebuffer data at the moment. 

This does produce a blank png of correct size... Might just need coeficients for r g and  b added , to get proper colors and I don't know them. 

Cheers,
YU4EAX

ea, 2017-05-28:
I should add, previous code requires pypng lib to be available, not sure what's available by default across all python installations that could be used in place of it.

ea, 2017-05-28:
Sorry, was editing on the go... Here's a cleaned up function:
```
def screenshot(dfu, filename="screenshot.png"):
  """Reads the LCD framebuffer"""
  rows = []
  for y in xrange(0,128):  
    buf = dfu.read_framebuf_line(y)
    row = []
    for i in range(0,len(buf)-1,2):
        uint16 = struct.unpack("H",buf[i:i+2])[0]
        b = ((uint16  >> 11 ) * 8 ) 
        g = (((uint16 & 0x07e0) >> 5 ) * 4)
        r = (((uint16 & 0x001f)) * 8)
        pixel = [r,g,b]
        row.append(pixel)
    rows.append(row)
  import png
  image = png.from_array(rows,"RGB",info=dict(height=128))
  image.save(filename)
```

Wolf-DL4YHF, 2017-05-28:
Many thanks, I will try this later.
The native colour format in the framebuffer is BGR565 ("blue bits" 
15..11, "green bits" 10..5, red in bits 4..0) so some bit-fiddling in 
the Python script is required. Also the USB transfer occasionally loses 
a line of pixels for obscure reasons.
Guess I need to echo the tile number, or the tile coordinates back in 
the USB response so the script can simply request it a second time if 
necessary.

Cheers,
   Wolf .

Wolf-DL4YHF, 2017-06-10:
With some help from K6BSD, I have a configuration for the FSMC (static memory controller) timing registers now, that can *reliably* read from the graphic framebuffer. It turned out that read-accesses to the LCD controller (which btw is an HX8302-A) are incredibly slow - much slower that I had assumed "in my wildest dreams", but for a screenshot via USB, it doesn't matter much.
There is even a modified Python script that saves the data as a valid bitmap file, together with an ugly hard-coded bitmap file header.  This way we can avoid new dependencies in the basic md380tools scripts (especially from modules that may not be available for all target platforms).
I will post the modified files (lcd driver + usb.c) in a few days. The code needs some cleanup after a lot of 'trial-and-error' experimentation with the FSMC settings.

DL2MF, 2017-12-30:
@Wolf-DL4YHF Hi Wolf, AFAIK this already has been realized months ago by a PR from your side? Could this issue be closed?

Wolf-DL4YHF, 2017-12-31:
Yes, the screenshot utility is in there since a couple of months, along 
with a python (wxpython) scipt that can periodically read out screens 
via USB (poor man's remote control, including keys sent from PC to MD380).
The script md380remote.pyw used to by in my temporary branch, not sure 
if it made it into the official repository.

Cheers any HNY,
   Wolf .

travisgoodspeed, 2017-12-31:
I added a help listing for the command.  Reopen this issue if it breaks in some situations.

commit 215f713f (2017-12-31): Adds a screenshot command to md380-tool.  Close #754.
closed 2017-12-31 by travisgoodspeed