travisgoodspeed/md380tools #761: Scan mode disabled after exiting red button menu

c5e3, 2017-05-22:
While being in scan mode,When entering the red button menu, scan mode is still on and the correct channels gets displayed.
After exiting the menu by pressing the red button again or exiting by navigating to 'exit' and pressing the green button, scan mode is disabled for any reason.

DL2MF, 2017-05-22:
Noticed the same here on my MD390/G, scan always stops when exiting the altmenu.

Additionally I would like to suggest, to move "red button" menu to another button. The red button is used for exit/back from every menu and also is nice to check the radio status (enable backlight). With the new upcoming feature of key repeatmode, there will be very often unintentional malfunction when exiting the factory menu.

I would prefer to activate the alt/appmenu by long keypress on green button or by press the hashkey.

Wolf-DL4YHF, 2017-05-22:
The reason is simple to explain but not easy to fix:

When exiting from the red button menu, the original firmware must be 
tricked to redraw the normal screen completely.
This is done by setting a global variable (channel number) to zero, 
which causes the Tytera firmware to refresh everything.
This has a similar effect like rotating the channel knob. Unfortunately 
this also seems to stop scanning.
Other methods didn't work here (including the trick with some 
GUI-opmode-something, which crashed the display, same as when trying to 
exit from the original netmon screens).

The code is in app_menu.c : Menu_Close(). That's the place where some 
kind of flag to start updating the main screen would be set.
I have deliberately left the non-functional method (with gui_opmode1 & 
2) in there.
Note: This stuff is tricky. Avoid calling stuff inside the original 
firmware, without carefully checking from where Menu_Close() is called, 
and from where the original firmware function is supposed to be called. 
Violating this rule leads to unstable behaviour (been there, tried a 
lot, before I decided to write the alternative menu and its alternative 
LCD driver. Less "surprises" when keeping the number of calls to the 
original firmware as low as possible.

Cheers,
   Wolf .

c5e3, 2017-05-22:
Thanks for the quick reply, Wolf!
An alternative LCD driver would be overkill. Since it is this complicated, we can live with this minor bug, I guess. I just thought, that there might be an easy fix, didn't expect it to be this complex.

Wolf-DL4YHF, 2017-05-22:
Well maybe it *is* easy, if we new how. Maybe it's just a bit somewhere in one of Tytera's variables im RAM that needs to be set, to force updating the screen ("Tytera's screen") completely.
Something like
   redraw_screen = 1;  // redraw everything on the main screen

But I'm afraid there isn't such a flag, and reverse-engineering the code to discover the relevant places in the disassembled original firmware is very time consuming. For me, 'forward engineering' (i.e. writing my own additions in C, or maybe Cortex assembler) is much easier, which is one of the reasons for beginning to develop the red-button menu (which does indeed have its own, tiny, new LCD driver).

About the "red menu key" : I wouldn't move this to another key. Like the "green menu key", it's similar to a function key. The two cursor keys between these two should keep their function (we may use them one day in the main screen to increment / decrement the frequency or the channel number). The keys in the 2nd to 4th row immediately open special-purpose screens, and I wouldn't like to change this principle now (it's already well established). The '*', '0', and '#' keys will sooner or later be needed for something else, so we really shouldn't occupy them yet.
Similar for '7', '8', '9' : When on the main screen, let's keep those keys for the future - don't use them for fixed special functions (like even more Netmon hotkeys), at least not yet.

DL2MF, 2017-05-23:
OK, like already discussed by mail, future keyb layout needs to respect further needs.

paulomartinspt, 2017-05-24:
`When exiting from the red button menu, the original firmware must be tricked to redraw the normal screen completely. This is done by setting a global variable (channel number) to zero, which causes the Tytera firmware to refresh everything.`

dumb question: why not call a reboot when exiting the red key menu? as long at it remains on the same channel, scan will start after reboot (i believe)

shelllanes, 2017-05-24:
the dmr marc had move to a dif website there is the link for it http://www.va3xpr.net/dmr-marc-worldwide-net/ plz fix it in the glv software it keep time out when it try to get the date base 
this is the new web site for it http://dmrworldwidenet.blogspot.ca/ plz fix it asap thank you

Wolf-DL4YHF, 2017-05-24:
Wrong topic, wrong group ?

Wolf-DL4YHF, 2017-05-24:
It would take inacceptably long to reboot, and all temporary changes to 
variables in RAM (i.e. zone number) would get lost immediately after 
setting them. So this is not an option.

travisgoodspeed, 2017-05-25:
Tagging this as "won't fix" until a clean solution presents itself.