travisgoodspeed/md380tools #763: Red Button Menue changing Zone - GPS Version not stored

DL9SK, 2017-05-22:
Version 2017-05-20 bc3586b GPS MD390

When changing the zone with/up down button in the red button menue it will not be stored after leaving the menue. 
This happens only with the gps version. I have both MD-390 version with and without gps here. 
The non gps version works correct. Zone was stored after leaving red button menue. 
Ii would be great if it will be fixed.

Wolf-DL4YHF, 2017-05-22:
Hi Ralf,

I would have implemented that myself, but I have no GPS to find out the 
addresses where the GPS-variant stores the zone information (in RAM, 
since the zone-number is not stored immediately in Flash by the "red 
button menu" yet). I can make a guess by looking at the disassembly, but 
it's an extremely tedious and time-consuming effort.
If anyone with a GPS-radio can find or check that address, I can quickly 
add them in the firmware for your radio.
In the non-GPS variant, I already found the address (and added it as a 
symbol in codeplug.h) :
     CODEPLUG_RAM_ADDR_ZONE_NUMBER_STRUCT = 0x2001E57C .

For the GPS variant, the address will be *similar* but not necessarily 
the same.
If you think you know it, you can watch the address in the tiny 
hexadecimal viewer:
- open the new menu with the red button
- press '1' to scroll down quickly to 'test / setup'
- press green key to enter the submenu
- press '4' to scroll down quicky to 'HexMon, and again the green key
You can type over the address shown there, hex digits A..F can be 
reached by presssing the white arrow up

The hex monitor shows a few hundred bytes, frequently updated. The 
current zone number is the fourth byte of a five-byte structure, most of 
them will show "FF" (don't ask me why, ask Tytera..) so it's quite easy 
to find. Try a few different zones in the original Tytera menu, and if 
the hex monitor always shows the expected value - bingo, got it.

The same principle can be used to find other items in RAM, CPU-Flash, or 
SPI-Flash too. But that's OT here, one day I will properly document the 
hex monitor..

Cheers,
   Wolf DL4YHF .

DL9SK, 2017-05-23:
Hi Wolf, 

I checked it. It should be 2001E643

I send you the Hex Screenshots per Mail

73 Ralf DL9SK

DL9SK, 2017-05-23:
Here the screenshot
[image]

Wolf-DL4YHF, 2017-05-23:
Hi Ralf,

Many thanks - if the currently active zone was number 7 when you made 
the screenshot, the start address of the "five-byte-struct" would be at 
2001E64C : Three bytes with FF, then the zone number, followed by a 
fourth FF.
As expected, the data are as in the non-GPS firmware (where the 
structure begins at 2001E57C), only the address is different. The 
address difference (0x00D0) also sounds plausible.

I will have a look at this tomorrow; maybe we simply try with a 
preliminary firmware for the GPS radio that I can upload to my website 
(binary file for downloading with the Tytera firmware updater, or with 
the md380-tools).

73,
   Wolf DL4YHF .

Jumbotroll, 2017-05-23:
Tested on MD390 GPS and the Zone is now stored. Great job!

Wolf-DL4YHF, 2017-05-23:
Thanks for testing, and again to Ralf for searching and finding the address.
Going to merge these three new lines of code, since they won't do harm.
The Travis CI server seems to be working again, all tests passed (also 
with the clang compiler).

Cheers and good night from Germany,
   Wolf DL4YHF .

DL9SK, 2017-05-24:
Now it works perfect. Many thanks for fast fixing to Wolf. 

73 Ralf DL9SK

closed 2017-05-24 by DL9SK