travisgoodspeed/md380tools #764: Talker Assist rarely displays (MD390GPS)

kb8fk, 2017-05-23:
First noticed on 19 May. Even if TA only is selected, I only get User DB display.

kb8fk, 2017-05-23:
Sorry, forgot to mention that it was working more consistently before that.

ok2it, 2017-05-23:
Can you confirm you are seeing Talker Alias data comming through frequently on screen #4 ??
This could be related to hotspot or repeater not sending TA at all or less frequently.
I have seen simillar behaviour while working over mmdvm rptr with signals on the edge.

kb8fk, 2017-05-23:
Yes, I see the data. I have found that if I change to another channel then back during reception I get the TA display. I am running a DVMEGA/rpi3 hotspot with Mmdvmhost.

DL2MF, 2017-05-23:
Talker alias display needs complete RX/decode of the TA header data, this may depend on your frequency offset and hotspot drift. The recent and current release (pending PR) doesn't contain any changes for TA. I have it myself in use and today received many TA frames via my local repeater without any issues.

kb8fk, 2017-05-23:
OK, I'll check my hotspot, Mmdvmhost setup and prior firmware. If I discover anything new, I'll report back. Thanks for the feedback and for all the great work you all are doing!

closed 2017-05-23 by kb8fk