travisgoodspeed/md380tools #774: updatedb errors

jimlaughlan, 2017-06-03:
Any ideas for cause?

2: ID->2048 Country->NL Address->hoseline.pd0mz.nl
Traceback (most recent call last):
  File "get_special_IDs.py", line 40, in <module>
    response = urlopen(req, timeout=20)
  File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python2.7/urllib2.py", line 429, in open
    response = self._open(req, data)
  File "/usr/lib/python2.7/urllib2.py", line 447, in _open
    '_open', req)
  File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 1228, in http_open
    return self.do_open(httplib.HTTPConnection, req)
  File "/usr/lib/python2.7/urllib2.py", line 1201, in do_open
    r = h.getresponse(buffering=True)
  File "/usr/lib/python2.7/httplib.py", line 1136, in getresponse
    response.begin()
  File "/usr/lib/python2.7/httplib.py", line 453, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python2.7/httplib.py", line 409, in _read_status
    line = self.fp.readline(_MAXLINE + 1)
  File "/usr/lib/python2.7/socket.py", line 480, in readline
    data = self._sock.recv(self._rbufsize)
socket.error: [Errno 104] Connection reset by peer
Makefile:41: recipe for target 'special.tmp' failed
make[2]: *** [special.tmp] Error 1
make[2]: Leaving directory '/home/seamus/md380tools/db'
Makefile:12: recipe for target 'update' failed
make[1]: *** [update] Error 2
make[1]: Leaving directory '/home/seamus/md380tools/db'
Makefile:76: recipe for target 'updatedb' failed
make: *** [updatedb] Error 2

travisgoodspeed, 2017-06-03:
Looks like the server is down, but we ought to gracefully fail for all but the main database.

jimlaughlan, 2017-06-03:
Issue resolved. Thanks :)

closed 2017-06-03 by jimlaughlan